JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #1: 大佬能否出一期用Entware安装V2ray的教程？

lnedpaul, 2018-05-24:
RT，wget不知道是install.sh还是v2ray的包，我是mips的路由器。

JACK-THINK, 2018-05-31:
不好意思，v2ray我没有研究过。你先在网上找找其他资料吧。

JACK-THINK, 2018-07-21:
v2ray无法做到透明代理，因为华硕官方固件和梅林原版固件不支持tproxy，如果要开启，需重新编译固件，这和本系统开发初衷相违背。如果仅需要路由器部署v2ray客户端，PC浏览器上用SwitchyOmega，我有相关方案。 @lnedpaul

lnedpaul, 2018-07-25:
感谢大佬回复，应该也解决了很大一部分人的疑问吧。

edwinhuish, 2018-08-12:
我代码小白，不过估计这个有参考意义。

https://github.com/XX-net/XX-Net/issues/11028

JACK-THINK, 2018-08-25:
@edwinhuish 我看了XX-net的文档，它也是socks代理，和我提供的v2ray插件一样，都需要浏览器使用SwitchyOmega，做不到透明代理。其实，仅透明代理tcp流量是可以的，不过udp是大问题，尤其是DNS解析！有不少方案使用了chinaDNS+dnsmasq的方案，不过这个性能和稳定性太差了，完全不能和openvpn等相提并论。真有透明代理需求，尽量使用OpenVPN吧，使用单域名映射多服务器方案，可实现长久的稳定连接，不会被墙。

closed 2018-10-29 by JACK-THINK

chenwei0930, 2022-02-09:
> @edwinhuish 我看了XX-net的文档，它也是socks代理，和我提供的v2ray插件一样，都需要浏览器使用SwitchyOmega，做不到透明代理。其实，仅透明代理tcp流量是可以的，不过udp是大问题，尤其是DNS解析！有不少方案使用了chinaDNS+dnsmasq的方案，不过这个性能和稳定性太差了，完全不能和openvpn等相提并论。真有透明代理需求，尽量使用OpenVPN吧，使用单域名映射多服务器方案，可实现长久的稳定连接，不会被墙。

我现在使用SwitchyOmega，不需要透明代理。现在安装了entware的v2ray，作为client，连接到我外部的v2ray服务。SwitchyOmega配置后，浏览器访问google，v2ray client有日志输出，没有报错，但是浏览器却显示无法访问。不知道那里出问题了。不知道您有没有方案，我参考一下。