JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #2: home目录重复挂载

cshlcsh, 2018-07-06:
两个USB设备一共五个分区，每一个分区挂载上的时候都会把SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER脚本触发一遍，于是home目录就会被挂载五遍，虽然没什么影响但是总觉得哪里不对。。

另外请教一下数学大神，我的aria2下载制定的路径所在文件夹总是会被挂载错误，比如label是G的分区，总是被挂载成了G(1)，然后就会引发很多错误。。。可以的话烦请大神帮忙鉴定一下～感激不尽！
[ERR LOG.TXT](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/files/2170000/ERR.LOG.TXT)

edwinhuish, 2018-08-12:
同样的问题，请问答案是什么？？

我需要挂一个U盘和一个移动硬盘，同样也是会将sda1挂载2次home。请问将U盘挂载到home的意义是什么?

```
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                42368     42368         0 100% /
devtmpfs                127752         0    127752   0% /dev
tmpfs                   127856      1136    126720   1% /tmp
/dev/mtdblock4           64256      1828     62428   3% /jffs
/dev/sda1             59581484    670944  55883960   1% /tmp/mnt/ASUS_ROUTER
/dev/sda1             59581484    670944  55883960   1% /tmp/home
/dev/sdb1            307663992  73380436 218655116  25% /tmp/mnt/Cloud
/dev/sda1             59581484    670944  55883960   1% /tmp/home
```

edwinhuish, 2018-08-12:
通过强大的Google，已经找到楼主的链接。

http://koolshare.cn/thread-138615-5-1.html

edwinhuish, 2018-08-12:
实践结果，AC1900在关机并不会执行script_bootloader_usb_umount，所以问题仍然无法解决。

edwinhuish, 2018-08-12:
而且貌似插入两个盘，所有代码都会执行两次？因为list_of_user_custom_scripts也执行两次了。请问能否修正这个BUG？