JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #7: ankiserver同步时高内存占用导致死机，有什么解决方案

yzswk, 2018-11-09:
我的是asus 68u,安装后，ankiserver无法自动启动，每30秒被启动一次，但很快进程就没有了。用ankiserverctl.py debug执行，服务会稳定存在，同步时大概传输到17M数据的时候，客户端会提示同步出错，这时服务器端虚拟内存占用很高，显示有300-500M，CPU占用100%，基本处于死机状态，需要手动kill掉进程才会恢复正常，请问可能是什么原因，怎么解决？

JACK-THINK, 2018-11-11:
我在anki_server的Readme中作了说明，目前不能开机自动启动。anki_server的主程序是python可执行程序，使用本系统提供的两种开机自动启动方法都会出现“很快进程就没有了”这种情况。如何解决我也不清楚，我不会用Python。希望有了解Python程序运行机制的朋友提出解决方案。

la8896, 2018-12-09:
您好,請問shadowsocks-server能否增加obfs-server

yzswk, 2018-12-19:
按anki server官方安装方法，用supervisor是可以开机正常自启动的，服务稳定存在，不知可不可以研究一下

nonwill, 2019-04-08:
该问题已解决，参考我提交的[pull request](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/pull/10)

JACK-THINK, 2019-09-21:
@nonwill 
我已重写anki_server插件，现在已经实现一键安装，免配置直接使用。谢谢您的pull request

closed 2019-09-21 by JACK-THINK