JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #8: 关于配置SS信息

weiyideai520, 2019-02-03:
小白求问
配置SS的三个文件信息 config_local，config_redir，config_tunnel，里面的本地端口是要都一样吗？
还是要不一样？

关于SS加密方式支持哪些呢？我填了chacha20，好像不起作用。但是我的服务端又不提供 你填写的默认加密方式

weiyideai520, 2019-02-15:
哥，这个SS实在不知道怎么设置，麻烦出个详细的教程。
那种透明代理的教程

weiyideai520, 2019-02-16:
搞了半天，终于把SS跑起来了
需要删掉
·····
"mode":"tcp_and_udp",
"plugin":"obfs-local",
"plugin_opts":"obfs=http;obfs-host=www.bing.com"
·····

但是透明代理还是没有整明白，大哥能发下的配置或者详细的教程，实在是搞不来啊~

JACK-THINK, 2019-03-05:
1. config_local，config_redir，config_tunnel，里面的local_port肯定是不能一样的
2. 配置文件默认带混淆，看来你的服务端不支持混淆
3. 你把config_local，config_redir，config_tunnel配置文件改好，再把overture的配置文件改好，直接在monit页面中开启SS_TRANSPARENT_PROXY_OVERTURE就可以透明代理了

jixihbz, 2019-04-14:
大师，请问在哪里查看和管理安装的插件程序

JACK-THINK, 2019-05-02:
@jixihbz 
用浏览器访问：
`<路由器IP>:35490`
就行了

jixihbz, 2019-05-02:
我按照您的教程安装好后，发现一个程序都没装上，是什么原因。有没有更详细的教程啊。

JACK-THINK, 2019-05-03:
@jixihbz 

如果你用浏览器访问：
`<路由器IP>:35490`
进入了monit管理界面，就说明你装上了大部分内容。如果没有，你需要提供日志。只要能够执行安装程序（全部安装完毕需要很久），就不太可能“一个程序都没装上”。

z4jst, 2019-07-21:
查了一下文档，按我的理解来说。
ss-local 就可以开启代理，等同于 v2 clash 等软件
启用 ss-redir 就可以满足 透明代理，可以用 clash、redsock 之类的替代
ss-tunnel 和 overture 功能重复
ss-serve 基本用不到
所以还是希望大佬可以将 overture 或者 dnsforwarder 等软件独立出来。另外可以希望尽快兼容 clash 啦。
谢谢带来这么好的软件 :D

JACK-THINK, 2019-07-24:
@z4jst 
你安装本系统中的shadowsocks和v2ray两个软件试试就能发现它们的重大区别：
1. shadowsocks是为了迎合低性能CPU和小内存的路由器；v2ray占用内存太高。
2. v2ray看起来是代理，但是在本系统中这不是它的重点。在这里它是被当作路由器+代理+VPN使用的。你看看它的配置文件就懂了，这是shadowsocks做不到的。
3. 关于ss-tunnel和overture。它们看似重复，实则不是。ss-tunnel不支持obfs混淆，但是ss-local支持。因此我专门用overture+ss-local来代替了ss-tunnel的功能。
4. ss-server有很多人当VPN用。适用于不能运行v2ray的路由器。
5. clash已在开发计划中。

closed 2019-12-10 by JACK-THINK