JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #12: stage 2 failed,  HOME HAS NOT BEEN MOUNTED

wowliuzj, 2019-07-13:
你好，自动安装执行到stage 2 时报错，错误信息显示 HOME HAS NOT BEEN MOUNTED，请问这个怎么解决？谢谢！

wowliuzj, 2019-07-13:
不好意思，机型是RT-N66U，插入的U盘是8G大小的

JACK-THINK, 2019-08-18:
@wowliuzj 
RT-N66U过于古老，我没有同型号的开发机，无法确定问题到底出在哪里。不过，你可以加我的Telegram，有时间我可以在线帮你调试

wowliuzj, 2019-08-18:
我也准备换一个路由，有什么型号推荐的吗？谢谢

JACK-THINK, 2019-08-19:
@wowliuzj 
1. 由于802.11AX（即WIFI 6）正式标准将于2020年上半年发布，因此在这个时间点买路由器非常尴尬。考虑到新品上市时间，建议1年半以后买WIFI 6协议的路由器
2. 如果实在无法忍受，并且资金充足，买华硕AX88U，可使用5年+（不太建议，成本太高）
3. 如果实在无法忍受，并且资金不充足，先买华硕AC86U，过1-2年卖掉，换AX新品（此方案相对可行）

JACK-THINK, 2019-08-19:
@wowliuzj 
目前经测试，本系统在AC68U（梅林固件）和AC86U（原厂固件）上运行良好。我没有和您同型号的开发机，无法确定问题到底出在哪里。不过，你可以加我的Telegram，有时间我可以在线帮你调试

可能的修复方法：
1. 报错后，你看看目录/tmp/mnt/ASUS_ROUTER/home/root/是否存在
2. 如果不存在，拔掉U盘，重启路由器，查看是否存在目录/tmp/home/root/
3. 查清后回复我上述结果

wowliuzj, 2019-08-19:
感谢回复，出差在外地机器目前不在身边，要等2个星期左右才能回去，回去了我测试过以后给您回复。

closed 2019-10-18 by JACK-THINK