JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #14: 在线安装建议先优先安装entware并升级wget后才进行 wget https://api.github.com

edwinhuish, 2019-08-01:
如下图，由于梅林自带的wget版本为1.16，不支持新版的TLS，导致每次到 wget https://api.github.com 都会出错。

建议先优先用 https://raw.githubusercontent.com，或者其他国内cdn先安装 entware并升级wget后才进行下面的一系列操作。

[image]

JACK-THINK, 2019-08-18:
@edwinhuish 您用的梅林版本是什么？我的梅林开发机是RT-AC68U，最新固件384.13-0中，wget版本是1.20，一切都能正常使用，不能复现您提到的问题。

[image]

edwinhuish, 2019-08-18:
今天升级到了384，确实是 1.20 的了，之前是 380 的，wget 是 1.16.

另外，之前Koolshare的梅林后台显示CPU占用经常双100%，monit 查看却是10%以下，通过各种 top， atop， htop 查看后，貌似是U盘读写繁忙所致。不确定是否 Syncthing 读写频繁所致。

但今天刷了原版的 384 的梅林后，U盘读写正常了，CPU占用也正常了。。。。

edwinhuish, 2019-08-18:
已确认是 SYNCTHING 的问题，刚刚添加了一个新的同步文件夹之后，AC68U 的后台再次双100%，

用 atop 查看后结果如下，sda 为 U盘， sdb 为硬盘。

[image]

JACK-THINK, 2019-08-20:
@edwinhuish 
不同于transmission这种专为低性能设备设计的工具，syncthing不是专为低性能设备而设计的，它的加密和文件校验是非常吃CPU性能的。因此，只要有文件同步，路由器CPU占用就会飞升，这很正常

本系统提供的syncthing组件就是syncthing官方提供的预编译版，你可以试试向syncthing官方反馈此问题，看看他们能否改进

edwinhuish, 2019-08-20:
非常感谢你的回复。

JACK-THINK, 2019-08-20:
@edwinhuish 
已新增samba回收站功能。如果你不想重装本系统，则修改文件/opt/etc/samba/smb.conf，在文件尾追加如下内容（最后的空行必须有），重启samba服务即可生效。回收站目录是/home/root/.recycle/
```shell
        #
        # Recycle Configuration
        vfs objects = recycle
        recycle:keeptree = yes
        recycle:versions = yes

```

edwinhuish, 2019-08-22:
@JACK-THINK 促使我不用你这个 samba， 改为折腾华硕原带的samba，主要是因为你这个samba不知道为什么在小米电视上无法通过网上邻居查看路由目录。试过好几个软件都是如此，只有 ES文件浏览器能够识别到，但是ES文件浏览器没有为电视操控优化界面。

请问是否 samba 还分为好几种模式？

JACK-THINK, 2019-08-22:
@edwinhuish 
由于安全问题（详见勒索病毒），我的SAMBA插件提供的最低协议版本是smb2，同时要求客户端必须支持smb2。电脑能看是因为Win10默认支持并启用smb2，电视不能看是因为它不支持smb2。正是因为梅林固件中的samba不能关闭smb1，所以我才开发了此插件（顺便集成了提速功能）

closed 2019-08-22 by JACK-THINK

edwinhuish, 2019-08-22:
谢谢 @JACK-THINK ，请问具体是什么问题？我这个samba仅在自己家里使用，应该没有什么问题吧？

JACK-THINK, 2019-08-22:
@edwinhuish 
就是勒索病毒会使用CIFS/SMB协议自动感染局域网内全部计算机。如果禁用了CIFS/SMB协议，会有所缓解，但也不是彻底解决问题。至于勒索病毒的原理和传播方式，你自己百度

Windows 10现在都是默认禁用CIFS/SMB协议的，转而使用SMB2协议。即使你仅在家中使用，如果有任何一台设备感染了勒索病毒，开启CIFS/SMB协议都可能导致局域网内全部计算机被感染，带来的损失会非常大（勒索病毒是可感染Windows，Linux和MacOS甚至还有Android的全系统病毒），因此，你最好禁用CIFS/SMB协议