JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #17: 安装最新11.0版出现的问题

lnedpaul, 2019-08-18:
`(entware_install): 1340 ***** INSTALLATION HAS BEEN COMPLETE *****
(entware_install): 1340 ***** COPY HOME TO USB FLASH DRIVE *****
(entware_install): 1340 ***** HOME HAS BEEN COPIED *****
(entware_install): 1340 ***** CREATE DIRECTORY /tmp/mnt/ASUS_ROUTER/www *****
(entware_install): 1340 ***** /tmp/mnt/ASUS_ROUTER/www HAS BEEN CREATED *****
(install): 1146 ***** HOME HAS NOT BEEN MOUNTED *****
***** STAGE 2: FAILED *****
`
发生了问题
AC66

JACK-THINK, 2019-08-19:
目前经测试，本系统在AC68U（梅林固件）和AC86U（原厂固件）上运行良好。我没有和您同型号的开发机，无法确定问题到底出在哪里。不过，你可以加我的Telegram，有时间我可以在线帮你调试

可能的修复方法：
1. 报错后，你看看目录/tmp/mnt/ASUS_ROUTER/home/root/是否存在
2. 如果不存在，拔掉U盘，重启路由器，查看是否存在目录/tmp/home/root/
3. 查清后回复我上述结果

lnedpaul, 2019-08-24:
拔去之前没有此文件夹，拔去重启路由器后就有了

lnedpaul, 2019-09-14:
下载再次安装12.0报错信息是timezone

JACK-THINK, 2019-09-21:
@lnedpaul 
在路由器设置中，把时间服务器设置为`ntp.aliyun.com`可解决此问题

lnedpaul, 2019-09-21:
> @lnedpaul
> 在路由器设置中，把时间服务器设置为`ntp.aliyun.com`可解决此问题

感谢，我试试看

closed 2019-10-18 by JACK-THINK