JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #18: aria2 严重占用CPU 和 IO

edwinhuish, 2019-08-25:
关闭所有插件后，CPU的占用回落到个位数，但一旦开启aria2，CPU占用飙升到80-90%，对比之下，甚至比syncthing还占 CPU 和 IO， 严重怀疑是BT下载惹的祸，导致后台偷偷当BT服务器了.。。

通过 AriaNg 修改下面配置，显示修改成功，但是一旦重启 Aria2 和 AriaNg, 这些修改的配置就会被还原？ 尽管我已经将 ``` /script_bootloader/usr/aria2/etc/aria2.conf ``` 设为 777。。。。

请问如何解决此问题？？

[image]

edwinhuish, 2019-08-25:
已通过手动修改``` /script_bootloader/usr/aria2/etc/aria2.conf ``` 配置文件，并移除之前下载的BT文件，CPU占用目前仅仅为个位数，暂时未测试下载新的BT文件，后续测试再对比。

JACK-THINK, 2019-08-28:
@edwinhuish 

1. BT下载就是吃CPU和IO。即使是i5-6600这种级别的CPU在满速BT下载时，CPU占用也会达到70%，IO占用达到100%

2. 通过AriaNG修改的配置无法保存到`/script_bootloader/usr/aria2/etc/aria2.conf`，[参考](https://github.com/mayswind/AriaNg/issues/337)

edwinhuish, 2019-08-28:
@JACK-THINK 出现问题的不是下载时的CPU和IO占用，时下载后还持续占用。我已经修改 aria2.conf，后面会持续观望资源占用情况

closed 2019-08-28 by edwinhuish

JACK-THINK, 2019-08-29:
@edwinhuish 

我在RT-AC68U上做了测试：在下载期间，内存和IO占用经常到90%以上，此时下载进度卡住，等待一段时间，会自动恢复。下载、校验完毕后，内存和IO占用可以被释放，恢复到下载前的状态。综上，不能复现你所描述的情况