JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #19: V12.0 - Monit Web Management Page

PC-Pilot, 2019-09-15:
V12.0

Unable to access Monit Web Management Page - Login Fails. No issue with V11.0 or previous?

Tried U/Name: admin P/Word: monit as in previous versions  also  U/Name & P/Word for Router Access (Asus RT-AX88U - router firmware Asuswrt Merlin 384.13)

Also repeated install on different Flash drive, same result.

From V12.0 Monit ReadMe:

Using the Web Management Page

1. Enter 'http://router address: 35490' in the browser address bar ! (step1) (.. /.. /..

/Documents-Assets/monit/web/step1.png)

2. Enter your username and password in the pop-up dialog box
User name: Router management page login username

From V11.0 Monit ReadMe:

Using the Web Management Page

1. Enter 'http://router address: 35490' in the browser address bar ! (step1) (.. /.. /..

/Documents-Assets/monit/web/step1.png)

2. Enter your username and password in the pop-up dialog box
User name: 'admin'
>

Password: 'monit' ! (step2) (.. /.. /..

/Documents-Assets/monit/web/step2.png)

Have the login details changed?  Or is there an issue??

PC-Pilot, 2019-09-18:
UPDATE

I have prepared a fresh Flash Drive and installed yesterdays V12.1 release (along with V13.0 Addons) and can report that the issue persists.

Previously I had no issue reaching the monit web interface (http://router address: 35490) with username 'admin', password 'monit' (without the quotes). I have also again tried using the router's access username & password (including separately containing the password in "double quotes" as it contains # character alongside numbers/alphanumeric) I can get none of the combinations to work.....

Any suggestions??

PC-Pilot, 2019-09-18:
UPDATE 2

Further investigation lead me to look in to the certification process and I observed that the Monit configuration file "monitrc" suggests that for secure access SSL should be enabled and offers a path  to the server certificate at the following location:  /etc/ssl/certs/monit.pem.  

A global router file search using WinSCP could locate no instances of "monit.pem" which might offer an explanation for the issue?

JACK-THINK, 2019-09-21:
@PC-Pilot 

1. SSL should do nothing with this problem. From Version 12.0&12.0, Monit turned to the username and password which are used to log in the management system from web instead of the default username `admin` and password ` monit`.

2. Log in the router with ssh, and check the values returned by `nvram get http_username` and `nvram get http_passwd` to see if they are identical to the values in line 160 in the file `/opt/etc/monitrc`. If they were identical, then you could log in the monit system with the two values returned.

3. If you want to use other pairs of username and password, you can modify the file '/opt/etc/monitrc' as below.
   ```shell
   157 set httpd port 35490 and
   158     use address 0.0.0.0  # only accept connection from localhost
   159     allow 0.0.0.0/0.0.0.0        # allow localhost to connect to the server and
   160     allow <username>:<password>      # require user 'admin' with password 'monit'
   ```
   > Change `<username>` to your username and `<password>` to your password and reboot your router, then the new pair of username and password will be in effect.

PC-Pilot, 2019-09-21:
@JACK-THINK 

Thank you for your detailed response and concise summary of the issue reported.

1.   I appreciate your clarification that the SSL should play no part. 

2.   I can confirm that the log in details at line 160 correspond with the nvram values although the passwd in both cases differs from the one I use - presumably as it is encrypted??  I can also confirm that it still does not function.

3.    The 'work around' means of varying the username/password via edit within 'monitrc' is much appreciated and should resolve loss of function. :)

4.    I suspect that you have identified the cause of the issue as my password contains the "#" character within the string.  Can you advise if there is there a means of enabling the router login access when special characters are present??  NB. I tried using double quotes around the password at the monit login prompt but this did not work either :(

JACK-THINK, 2019-09-22:
@PC-Pilot 

1. **Neither** the passwords returned by `nvram get http_passwd` nor the one in `/opt/etc/monitrc` is encrypted.

2. Special characters can cause this problem.
   
   Read these lines below from program `monit_install`
   ```shell
   # ADMINUSER: The administrator of the router
   ADMINUSER=$(nvram get http_username)
   #
   # ADMINPASSWORD: The administrator's password of the router
   ADMINPASSWORD=$(nvram get http_passwd)
   #
   /opt/bin/sed -i 's/allow admin:monit/allow '"${ADMINUSER}:${ADMINPASSWORD}"'/' ${PRIVATE_ETC}/monitrc
   ```
   I use command `sed` to substitute `admin` and `monit` with the values returned by `nvram get http_username` and `nvram get http_passwd`. Unfortunately, special characters like `#$;/:` have special function to `sed`. As a consequence, if the two returned values contain special characters like that, errors will come up.

   I'm still looking for some command to replace `sed`.

PC-Pilot, 2019-09-22:
Thank you for the clarification.

The described 'sed' limitation would likely explain the sequence of random alphanumeric characters reported in lieu of my password when running the command "nvram get http_passwd" and thus my assumption of encryption being applied.

I will look out for any future changes you are able to implement which can resolve the current "special characters" limitation and thus restore the intended router username/password functionality. 

In the meantime the suggested manual edit of  "monitrc" should will provide an adequate work around solution.

closed 2019-10-12 by JACK-THINK