JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #21: 关于 label_modifier 更改 U 盘卷标的问题

stuarthua, 2019-10-08:
### 设备&环境

* RT-AC1900P（网评与 RT-AC68U 基本相同）
* Version: 13.1&13.2

### 问题描述

请原谅我使用的路由器是 RT-AC1900P，知道并不完全兼容该项目提供的设备列表。但我还是描述一下问题吧，毕竟这两个路由器实在没有什么太大的差异（在固件方面）。

在执行 `label_modifier` 时，未能卸载 U 盘挂载导致修改 U 盘卷标失败，查看了一下代码

```shell
# label_modifier 第 13 行
USB_DEVICE_PARTITION="/dev/$(cat /tmp/usb.log | sed -E -n -e 's/.*\((sd[[:alpha:]]1)\):[[:space:]]+Success.*/\1/p')"
``` 

不知道是不是设备差异造成 `usb.log` 生成不一致，在 RT-AC1900P 上执行

```bash
$ cat /tmp/usb.log | sed -E -n -e 's/.*\((sd[[:alpha:]]1)\):[[:space:]]+Success.*/\1/p'

// 无结果

$ cat /tmp/usb.log

// 显示如下：
[usb_dbg: asus_usb_interface] (1-2:1.0): action=add.
[usb_dbg: asus_usb_interface] (1-2): action=add.
[usb_dbg: asus_usb_interface] (1-2): Fail to get usb node.
[usb_dbg: asus_usb_interface] (1-2:1.0): Not modem interface.
[usb_dbg: asus_sg] (sg0): action=add.
[usb_dbg: init_3g_param] 3G: Auto setting.
[usb_dbg: asus_sg] (sg0): modprobe cdrom!
[usb_dbg: asus_sd] (sda): action=add.
[usb_dbg: asus_sd] (sda): Got usb node: 1-2.
[usb_dbg: asus_sd] (sda): Got usb path: 2.
[usb_dbg: asus_sd] (sda): Got usb port: 1-2.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: asus_sd] (sda): Had 1 interface on Port 1-2.
[usb_dbg: asus_sg] (sg0): Success!
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: add sda 1-2 storage.
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: for sda.
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: sda: start to set usb_path...
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: add sda 1-2 storage: End.
[usb_dbg: asus_sd] (sda): Success!
```

暂时更改了一下 `label_modifier` ，以便正常执行

```
USB_DEVICE_PARTITION="/dev/sda"
```

### 希望

由于手头上没有其他路由器设备测试，不清楚是不是 `usb.log` 在不同设备上写法不同，提供一份 `usb.log` , 希望 `sed` 时可以顺便正则到这种写法吧

```shell
$ cat /tmp/usb.log

[usb_dbg: asus_usb_interface] (1-2:1.0): action=add.
[usb_dbg: asus_usb_interface] (1-2): action=add.
[usb_dbg: asus_usb_interface] (1-2): Fail to get usb node.
[usb_dbg: asus_usb_interface] (1-2:1.0): Not modem interface.
[usb_dbg: asus_sg] (sg0): action=add.
[usb_dbg: init_3g_param] 3G: Auto setting.
[usb_dbg: asus_sg] (sg0): modprobe cdrom!
[usb_dbg: asus_sd] (sda): action=add.
[usb_dbg: asus_sd] (sda): Got usb node: 1-2.
[usb_dbg: asus_sd] (sda): Got usb path: 2.
[usb_dbg: asus_sd] (sda): Got usb port: 1-2.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: asus_sd] (sda): Had 1 interface on Port 1-2.
[usb_dbg: asus_sg] (sg0): Success!
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: add sda 1-2 storage.
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: for sda.
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: sda: start to set usb_path...
[usb_dbg: set_usb_common_nvram] set_usb_common_nvram: add sda 1-2 storage: End.
[usb_dbg: asus_sd] (sda): Success!
```

stuarthua, 2019-10-08:
补充说明：

测试使用的是梅林固件，版本号是 [RT-AC68U_384.13_0.zip](https://sourceforge.net/projects/asuswrt-merlin/files/RT-AC68U/Release/RT-AC68U_384.13_0.zip/download)

JACK-THINK, 2019-10-10:
@stuarthua 

谢谢反馈

根据您提供的`/tmp/usb.log`，路由器挂载U盘为`/dev/sda`，但是没有继续挂载分区，所以没有显示`[usb_dbg: asus_sd] (sda1): Success!`。您是否方便告知在安装本系统前，U盘是什么状态？有几个分区？是什么格式（fat/exfat/ntfs/...）？谢谢。

RT-AC1900P应该是受支持的。出现这个问题，应该是本系统的Bug。编写时，未考虑到这种情况，因为我用的U盘是只有1个分区，格式为ext3的U盘。可能您的U盘上没有分区，或者有路由器不能识别的分区格式，才会出现此问题。

后续会修正此Bug。

stuarthua, 2019-10-10:
@JACK-THINK 

感谢提醒，问题的原因 `U 盘没有分区或路由器不能识别分区格式`。

重新检查了一下，之前格式化的方式错误，没有正确分区，所以才导致该问题。

stuarthua, 2019-10-10:
正如你所说，路由器挂载了 U 盘，但没有继续挂载分区。

### 操作环境

* Mac
* Homebrew

查看硬盘情况

```
$ diskutil list

/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:                 Apple_APFS Container disk1         250.8 GB   disk0s2

/dev/disk1 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +250.8 GB   disk1
                                 Physical Store disk0s2
   1:                APFS Volume Macintosh HD            125.2 GB   disk1s1
   2:                APFS Volume Preboot                 47.7 MB    disk1s2
   3:                APFS Volume Recovery                510.4 MB   disk1s3
   4:                APFS Volume VM                      3.2 GB     disk1s4

/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *4.1 GB     disk2
   1:                 DOS_FAT_32 UDISK                   4.1 GB     disk2s1
```

U 盘挂载为 /dev/disk2

错误的格式化 (导致 U 盘没有正确分区)

```
diskutil unmountdisk /dev/disk2
sudo $(brew --prefix e2fsprogs)/sbin/mkfs.ext3 /dev/disk2
```

纠正如下：

```
diskutil unmountdisk /dev/disk2
sudo $(brew --prefix e2fsprogs)/sbin/mkfs.ext3 /dev/disk2s1
```

JACK-THINK, 2019-10-10:
@stuarthua 

我已经修正此问题，现在将其更名为drive_modifier，可以自动分区并格式化U盘。版本号没有变，重新执行在线安装即可。

closed 2019-10-12 by JACK-THINK