JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #22: 无法格式化U盘

drmcachr, 2019-10-12:
路由器型号：AC1900P
固件版本：3.0.0.4.384_81049
现象：运行脚本安装的时候提示无法读取/dev目录

fdisk: can't read from /dev/
[: 0: unknown operand
(drive_modifier): 17729 ***** RE-PARTITION THE USB FLASH DRIVE *****
'/dev/' is opened for read only
fdisk: can't read from /dev/
(drive_modifier): 17729 ***** ERROR: CAN'T RE-PARTITION THE USB FLASH DRIVE *****
(install_online): 17521 ***** ABORT INSTALLATION *****

手动格式化U盘后发现ssh也无法访问/dev/sda目录

/dev# cd sda
-sh: cd: can't cd to sda

请问如何解决？

JACK-THINK, 2019-10-12:
@drmcachr 
拔掉U盘，重启路由器。路由器重新启动后，插入U盘，等10秒，执行下列代码：
```shell
cat /tmp/usb.log
```
然后把相关内容发给我。

drmcachr, 2019-10-12:
应该是USB_DEVICE那个sed执行的有问题，我把USB_DEVICE写死成/dev/sda就能执行下去了

JACK-THINK, 2019-10-12:
@drmcachr 
正常情况下，USB_DEVICE应从/tmp/usb.log取值，得到sda或者sdb什么的。根据你提供的信息，明显是sed表达式没能从/tmp/usb.log取出sda。这非常奇怪。因为这个表达式在我这里是没有问题的。方便的话提供一份/tmp/usb.log的文件内容给我，我分析分析。

drmcachr, 2019-10-12:
稍等一下，正在下载sbl_setup，91M的文件，只有2k的速度，请问一下有离线安装的选项吗？

JACK-THINK, 2019-10-12:
@drmcachr 
直接Ctrl+C结束掉，然后从Github上下载，解压至U盘根目录。把script_bootloader文件夹放在U盘根目录，然后执行`cd /tmp && /tmp/mnt/ASUS_ROUTER/script_bootloader/bin/install`进行安装。

drmcachr, 2019-10-12:
`[usb_dbg: asus_usb_interface] (2-2:1.0): action=add.
[usb_dbg: asus_usb_interface] (2-2:1.0): Not modem interface.
[usb_dbg: asus_usb_interface] (2-2): action=add.
[usb_dbg: asus_usb_interface] (2-2): Fail to get usb node.
[usb_dbg: asus_sg] (sg0): action=add.
[usb_dbg: init_3g_param] 3G: Auto setting.
[usb_dbg: asus_sg] (sg0): modprobe cdrom!
[usb_dbg: asus_sd] (sda): action=add.
[usb_dbg: asus_sd] (sda): Got usb node: 2-2.
[usb_dbg: asus_sd] (sda): Got usb path: 2.
[usb_dbg: asus_sd] (sda): Got usb port: 2-2.
[usb_dbg: asus_sd] (sda): wait 1 second for the printer module on Port 2-2.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: asus_sd] (sda): Had 1 interface on Port 2-2.
[usb_dbg: asus_sg] (sg0): Success!
[usb_dbg: set_usb_common_nvram] sda: start to set usb_path...
[usb_dbg: set_usb_common_nvram] add sda 2-2 storage: End.`

drmcachr, 2019-10-12:
[usb_dbg: asus_usb_interface] (2-2:1.0): action=add.
[usb_dbg: asus_usb_interface] (2-2:1.0): Not modem interface.
[usb_dbg: asus_usb_interface] (2-2): action=add.
[usb_dbg: asus_usb_interface] (2-2): Fail to get usb node.
[usb_dbg: asus_sg] (sg0): action=add.
[usb_dbg: init_3g_param] 3G: Auto setting.
[usb_dbg: asus_sg] (sg0): modprobe cdrom!
[usb_dbg: asus_sd] (sda): action=add.
[usb_dbg: asus_sd] (sda): Got usb node: 2-2.
[usb_dbg: asus_sd] (sda): Got usb path: 2.
[usb_dbg: asus_sd] (sda): Got usb port: 2-2.
[usb_dbg: asus_sd] (sda): wait 1 second for the printer module on Port 2-2.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: asus_sd] (sda): Had 1 interface on Port 2-2.
[usb_dbg: asus_sg] (sg0): Success!
[usb_dbg: set_usb_common_nvram] sda: start to set usb_path...
[usb_dbg: set_usb_common_nvram] add sda 2-2 storage: End.

JACK-THINK, 2019-10-12:
@drmcachr 
这个日志非常奇怪，你看看https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/issues/21帖子，他的路由器也是RT-AC1900P，/tmp/usb.log文件中包含`[usb_dbg: asus_sd] (sda): Success!`行，AC86U和AC68U的/tmp/usb.log文件中也包含`[usb_dbg: asus_sd] (sda): Success!`行。你换个U盘试试。我觉得不太对。

drmcachr, 2019-10-12:
没有，就挂了一个USB

drmcachr, 2019-10-12:
我用的原版固件，前几天才更新的81049，不知道和这个有没有关系

JACK-THINK, 2019-10-12:
@drmcachr 
这个真的很奇怪。建议你重启路由器，换个U盘再试试。如果还是这样，目前我也不清楚该如何解决。/tmp/usb.log是华硕固件自动生成的，它的生成规则我不太了解。

drmcachr, 2019-10-12:
等我把这个下载完了再插上给你看一下usb的log

JACK-THINK, 2019-10-12:
@drmcachr 
好的。

drmcachr, 2019-10-12:
顺带问一下，我想把https://github.com/xteve-project/xTeVe/releases这个作为一个addon执行，直接自建一个目录，然后放到bin目录就行了么？这个应该不需要安装

JACK-THINK, 2019-10-12:
@drmcachr 
不是。你看看使用手册吧，后面有提到插件的管理方式。

drmcachr, 2019-10-12:
[usb_dbg: asus_sd] (sda): Remove Storage on USB node 2-2.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: asus_usb_interface] (2-2:1.0): action=add.
[usb_dbg: asus_usb_interface] (2-2:1.0): Not modem interface.
[usb_dbg: asus_usb_interface] (2-2): action=add.
[usb_dbg: asus_usb_interface] (2-2): Fail to get usb node.
[usb_dbg: asus_sg] (sg0): action=add.
[usb_dbg: init_3g_param] 3G: Auto setting.
[usb_dbg: asus_sg] (sg0): modprobe cdrom!
[usb_dbg: asus_sd] (sda): action=add.
[usb_dbg: asus_sd] (sda): Got usb node: 2-2.
[usb_dbg: asus_sd] (sda): Got usb path: 2.
[usb_dbg: asus_sd] (sda): Got usb port: 2-2.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: asus_sd] (sda): Had 1 interface on Port 2-2.
[usb_dbg: set_usb_common_nvram] sda: start to set usb_path...
[usb_dbg: asus_sg] (sg0): Success!
[usb_dbg: set_usb_common_nvram] add sda 2-2 storage: End.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: set_file_integrity] delete /tmp/mnt/ASUS_ROUTER/.___var.txt.6.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: get_folder_list] There is no folder in /tmp/mnt/ASUS_ROUTER.
[usb_dbg: set_file_integrity] delete /tmp/mnt/ASUS_ROUTER/.__DreamCatcher_var.txt.6.
[usb_dbg: set_file_integrity] delete /tmp/mnt/ASUS_ROUTER/.__folder_list.txt.9.
[usb_dbg: asus_sd] (sda): Success!
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.

这回貌似有了

JACK-THINK, 2019-10-12:
@drmcachr 
有了就行了。那就可以正常安装了。

closed 2019-10-12 by drmcachr