JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #24: v2ray 开启透明代理后无法上网

z4jst, 2019-11-06:
大佬你好，请教一个问题：

问题：开启fwd后，路由器下所有设备无法上网
已安装插件：
V2RAY
ARIA2

fw_filter_INPUT_v2ray_proxy

V2RAY_TRANSPARENT_PROXY

DNSMASQ.D（未做任何配置）

复现步骤：
v2ray 已经调试成功，在关闭 fw_filter_INPUT_v2ray_proxy 的情况下，路由器中输入 curl -x 127.0.0.01:1080 http://www.google.com
可以返回正常的网页内容，网络连通，且路由器下所有设备正常上网。

在monit文件夹下，开启 fw_filter_INPUT_v2ray_proxy.on ，monit状态全部OK，但路由器下所有设备无法联网。只能关闭 fwd 并重启路由器。

配置：
v2ray 的配置文件
inbound 中，22815-22818 端口均补充了 uuid 以及 ss password
outbound 中，补充修改了 v2ray client 的配置
v2ray -test config 返回配置正常

其他配置文件未作修改。

z4jst, 2019-11-07:
昨天重新安装了一下 fwd 和 v2，restore 了整个 asusbootloader。 依旧如下图所示，但是 fwd 中出现了新的问题。显示 not in a function。 

不开 fwd 时，访问均正常，走内网代理时也正常。

但是开启 fwd 后，出现下图所示报错，且所有设备无法联网。 

fwd 的配置文件使用的是默认的，未作修改。

[image]

z4jst, 2019-11-07:
[image]

JACK-THINK, 2019-11-07:
1. MONIT服务V2RAY和MONIT服务V2RAY_TRANSPARENT_PROXY不能同时启用。若要使用全局透明代理，需要先关闭MONIT服务V2RAY，再关闭防火墙FW_PORT_OPENING_V2RAY_PROXY，然后再启用MONIT服务V2RAY_TRANSPARENT_PROXY

2. 防火墙FW_PORT_OPENING_V2RAY_PROXY是配合 MONIT服务V2RAY一起使用的。它的功能是允许公网设备访问你的路由器，和科学上网毫无关系

3. 默认情况下，防火墙FW_PORT_OPENING_V2RAY_PROXY是禁用的。如果没有特殊需要，不要启用它

z4jst, 2019-11-07:
你好

我关闭了 V2RAY 和 FW_PORT_OPENING_V2RAY_PROXY
仅开启了 V2RAY_TRANSPARENT_PROXY
路由器下所有设备仍然无法上网

这个是否跟我 v2ray 的服务是 h2+tls 有关呢？

JACK-THINK, 2019-11-07:
@z4jst 
就是这个问题。配置文件`/opt/script_bootloader/usr/v2ray/etc/config_proxy.json`默认的外连方式是tcp，请按需修改该文件中`VPS_outbound`部分。

z4jst, 2019-11-07:
我关闭了 V2RAY 和 FW_PORT_OPENING_V2RAY_PROXY
仅开启了 V2RAY_TRANSPARENT_PROXY
路由器下所有设备仍然无法上网

这个是否跟我 v2ray 的服务是 h2+tls 有关呢？

JACK-THINK, 2019-11-08:
@z4jst 
这个插件开发的时候，v2ray对http/2+tls的支持还不完善，所以当时的配置文件采用的tcp模式。由于此配置文件长期未更新，所以现在还是tcp模式。如果你需要使用tls模式，请自行修改配置文件。也许以后我会把默认的tcp模式修改为http/2+tls模式

先提供一个功能一样的http/2+tls的模板，你按提示修改后替换原配置文件`/opt/script_bootloader/usr/v2ray/etc/config_proxy.json`即可
```
{
    "inbounds":
    [
        {
            "port":1080,
            "listen":"0.0.0.0",
            "protocol":"socks",
            "settings":
            {
                "auth":"noauth",
                "udp":true,
                "ip":"127.0.0.1",
                "userLevel":1
            },
            "tag":"LAN_inbound_socks_VPS"
        },
        {
            "port":2053,
            "listen":"0.0.0.0",
            "protocol":"dokodemo-door",
            "settings":
            {
                "address":"<上游DNS地址>",
                "port":53,
                "network":"tcp,udp",
                "timeout":0,
                "followRedirect":false,
                "userLevel":1
            },
            "tag":"DNS_local_dokodemo_VPS"
        },
        {
            "port":45154,
            "listen":"0.0.0.0",
            "protocol":"dokodemo-door",
            "settings":
            {
                "network":"tcp,udp",
                "timeout":30,
                "followRedirect":true,
                "userLevel":1
            },
            "streamSettings":
            {
                "sockopt":
                {
                    "tproxy":"redirect"
                }
            },
            "tag":"LAN_local_dokodemo_VPS"
        },
        {
            "port":22815,
            "listen":"0.0.0.0",
            "protocol":"vmess",
            "settings":
            {
                "clients":
                [
                    {
                        "id":"<使用vmess协议接入路由器所用的用户主ID>",
                        "level":1,
                        "alterId":64
                    }
                ],
[... 316 lines cut ...]
                "inboundTag":
                [
                    "WAN_inbound_shadowsocks_Router"
                ],
                "outboundTag":"Router_outbound"
            }
        ]
    },
    "policy":
    {
        "levels":
        {
            "1":
            {
                "handshake":4,
                "connIdle":300,
                "uplinkOnly":2,
                "downlinkOnly":5,
                "statsUserUplink":false,
                "statsUserDownlink":false
            }
        }
    }
}
···

z4jst, 2019-11-08:
你好，感谢你的回复
现在情况是：
1. 配置文件正常
2. 开启透明代理，设备无法联网
但是又没报错，所以不知道问题在哪。

monit 中运行的有
[image]

v2 配置文件的测试截图
[image]

配置文件以及截图如下：（已隐去关键信息）

{
    "inbounds":
    [
        {
            "port":1080,
            "listen":"0.0.0.0",
            "protocol":"socks",
            "settings":
            {
                "auth":"noauth",
                "udp":true,
                "ip":"127.0.0.1",
                "userLevel":1
            },
            "tag":"LAN_inbound_socks_VPS"
        },
        {
            "port":2053,
            "listen":"0.0.0.0",
            "protocol":"dokodemo-door",
            "settings":
            {
                "address":"8.8.8.8",
                "port":53,
                "network":"tcp,udp",
                "timeout":0,
                "followRedirect":false,
                "userLevel":1
            },
            "tag":"DNS_local_dokodemo_VPS"
        },
        {
            "port":45154,
            "listen":"0.0.0.0",
            "protocol":"dokodemo-door",
            "settings":
            {
                "network":"tcp,udp",
                "timeout":30,
                "followRedirect":true,
                "userLevel":1
            },
            "streamSettings":
            {
                "sockopt":
                {
                    "tproxy":"redirect"
                }
            },
            "tag":"LAN_local_dokodemo_VPS"
        },
        {
            "port":22815,
            "listen":"0.0.0.0",
            "protocol":"vmess",
            "settings":
[... 326 lines cut ...]
                "type":"field",
                "inboundTag":
                [
                    "WAN_inbound_shadowsocks_Router"
                ],
                "outboundTag":"Router_outbound"
            }
        ]
    },
    "policy":
    {
        "levels":
        {
            "1":
            {
                "handshake":4,
                "connIdle":300,
                "uplinkOnly":2,
                "downlinkOnly":5,
                "statsUserUplink":false,
                "statsUserDownlink":false
            }
        }
    }
}

JACK-THINK, 2019-11-09:
@z4jst 
请完全照抄我提供的配置文件，仅替换其中的`<中文>`部分。检测客户端的DNS配置是否为自动获取。用客户端的V2Ray客户端直接连接V2Ray服务器，检测服务器是否工作正常

如果还不行，执行下列操作：
1. 执行
```
cd /opt/etc/monit.d/
mv v2ray.on v2ray
mv v2ray_transparent_proxy_file.on v2ray_transparent_proxy_file
mv v2ray_transparent_proxy_prog.on v2ray_transparent_proxy_prog
mv dnsmasq.d.on dnsmasq.d
reboot
```
2. 路由器重启后，执行`/opt/script_bootloader/usr/v2ray/bin/v2ray_transparent_proxy_enable_file.service`
3. 1分钟后，检测路由器下设备能否科学上网。如还不行，继续下列操作
4. 开启透明代理，在路由器上执行`drill www.google.com`，并反馈输出
5. 开启透明代理，在路由器上执行`iptables -t nat -nvL`，并反馈输出
6. 开启透明代理，在路由器上执行`iptables -t mangle -nvL`，并反馈输出
7. 开启透明代理，在路由器上执行`cat /tmp/resolv.dnsmasq`，并反馈输出

z4jst, 2019-11-09:
你好，
1. 配置文件是完全照抄的。
我在开启 v2ray 的情况下，使用命令`curl -x socks5://127.0.0.1:1080 http://www.google.com`测试了 v2ray。
反馈
```
<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="en"><head><meta content="Search the world's information, including webpages, images, videos and more. Google has many special features to help you find exactly what you're looking for."...........
```
说明 v2 正常运行，配置文件一切正常。

2. 用您列的方法测试后，无法上网，全部命令反馈如下：
`drill www.google.com`
```
;; ->>HEADER<<- opcode: QUERY, rcode: NOERROR, id: 15706
;; flags: qr rd ra ; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0 
;; QUESTION SECTION:
;; www.google.com.      IN      A

;; ANSWER SECTION:
www.google.com. 300     IN      A       66.220.151.20

;; AUTHORITY SECTION:

;; ADDITIONAL SECTION:

;; Query time: 4 msec
;; SERVER: 192.168.1.1
;; WHEN: Sat Nov  9 22:41:02 2019
;; MSG SIZE  rcvd: 48
```
`iptables -t nat -nvL`
```
Chain PREROUTING (policy ACCEPT 141 packets, 10532 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    7   436 VSERVER    all  --  *      *       0.0.0.0/0            192.168.1.3         
   19  1172 V2RAY_TPROXY_TCP  tcp  --  *      *       192.168.0.0/16       0.0.0.0/0           

Chain INPUT (policy ACCEPT 134 packets, 10440 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 98 packets, 7876 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    1    60 V2RAY_TPROXY_TCP  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0           

Chain POSTROUTING (policy ACCEPT 12 packets, 750 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 ACCEPT     all  --  *      *       192.168.50.0/24      0.0.0.0/0            policy match dir out pol ipsec
  302 20729 PUPNP      all  --  *      eth0    0.0.0.0/0            0.0.0.0/0           
  186 12925 MASQUERADE  all  --  *      eth0   !192.168.1.3          0.0.0.0/0           
    8  2055 MASQUERADE  all  --  *      br0     192.168.50.0/24      192.168.50.0/24     
   85  6513 MASQUERADE  all  --  *      *       192.168.0.0/16      !192.168.0.0/16      

Chain DNSFILTER (0 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain LOCALSRV (0 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain PCREDIRECT (0 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain PUPNP (1 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain V2RAY_TPROXY_TCP (2 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/8           
    0     0 RETURN     all  --  *      *       0.0.0.0/0            10.0.0.0/8          
    0     0 RETURN     all  --  *      *       0.0.0.0/0            127.0.0.0/8         
    0     0 RETURN     all  --  *      *       0.0.0.0/0            169.254.0.0/16      
    0     0 RETURN     all  --  *      *       0.0.0.0/0            172.16.0.0/12       
[... 33 lines cut ...]
Chain POSTROUTING (policy ACCEPT 63316 packets, 69M bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain BWDPI_FILTER (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 DROP       udp  --  eth0   *       0.0.0.0/0            0.0.0.0/0            udp spt:68 dpt:67
    0     0 DROP       udp  --  eth0   *       0.0.0.0/0            0.0.0.0/0            udp spt:67 dpt:68

Chain V2RAY_TPROXY_UDP (2 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/8           
    0     0 RETURN     all  --  *      *       0.0.0.0/0            10.0.0.0/8          
  142 31590 RETURN     all  --  *      *       0.0.0.0/0            127.0.0.0/8         
    0     0 RETURN     all  --  *      *       0.0.0.0/0            169.254.0.0/16      
    0     0 RETURN     all  --  *      *       0.0.0.0/0            172.16.0.0/12       
  266 24366 RETURN     all  --  *      *       0.0.0.0/0            192.168.0.0/16      
   84 27403 RETURN     all  --  *      *       0.0.0.0/0            224.0.0.0/4         
    6  1464 RETURN     all  --  *      *       0.0.0.0/0            240.0.0.0/4         
  427 28722 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            MARK set 0x2333
```
`cat /tmp/resolv.dnsmasq`
```
#server=192.168.1.1
server=8.8.8.8
```

JACK-THINK, 2019-11-10:
@z4jst 
看了您的反馈，应该是`8.8.8.8`被运营商重定向了。按如下方式修改`/opt/script_bootloader/usr/v2ray/bin/fw_v2ray_transparent_proxy_enable.service`

1. 82行替换为如下代码
```
    TPROXY_MODULES_STATUS=false
```

然后再测试是否可以科学上网。还有，需要明确一下：您说的无法联网是指不能访问百度还是不能科学上网呢?

z4jst, 2019-11-10:
您好，感谢回复！
改了之后，测试果然成功了，现在可以正常上网了，谢谢。

之前说的无法联网是所有设备无法链接任意网络。

z4jst, 2019-11-10:
不过如果用 v2 的透明代理，分流或者规则，是不是就没办法使用了呢？

JACK-THINK, 2019-11-12:
@z4jst 
依据我自己的经验，任何使用白名单方式的分流代理软件，都可能导致部分墙外网站无法访问，或者部分墙内网站被解析到境外。至今没有完美的解决方案。而且，超长白名单很吃CPU，占用资源太高，不适合路由器使用。因此，我建议用户不要在路由器上搞分流

正确的解决方式如下：
1. 客户端安装代理软件，然后用代理软件指向路由器，需要科学上网时运行代理软件即可
2. 对于总在墙内和总在墙外的设备，使用dnsmasq.d服务，为它们指定不同的网关和DNS，让它们各自使用合适的网络环境
3. MONIT服务V2RAY_TRANSPARENT_PROXY开启后，访问任何网站都需要到墙外绕一圈，这是全局透明代理，日常使用可能导致国内网站认为你在境外。所以，还是使用chrome+switchyomega配合MONIT服务V2RAY比较好用

closed 2019-11-12 by JACK-THINK