JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #28: 安装的时候出现 ERROR: CAN'T DETECT THE USB FLASH DRIVE

Keenni, 2019-11-11:
Disk /dev/sda: 15.3 GB, 15376318464 bytes
255 heads, 63 sectors/track, 1869 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1   *           1        1870    15014912  83 Linux
admin@RT-AC88U-5AD0:/tmp# cd /tmp && wget -O /tmp/install_online --no-check-certificate "https://raw.githubusercontent.com/JACK-
THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/master/script_bootloader/bin/install_online" && chmod 777 /tmp/install_online && /tmp/i
nstall_online
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2019-11-12 00:38:42--  https://raw.githubusercontent.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/master/script_bootloader/bin/install_online
Resolving raw.githubusercontent.com... 151.101.88.133
Connecting to raw.githubusercontent.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7500 (7.3K) [text/plain]
Saving to: '/tmp/install_online'

/tmp/install_online             100%[=======================================================>]   7.32K  --.-KB/s    in 0s

2019-11-12 00:38:42 (15.7 MB/s) - '/tmp/install_online' saved [7500/7500]

Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2019-11-12 00:38:42--  https://raw.githubusercontent.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/master/script_bootloader/bin/prerequisite_checker
Resolving raw.githubusercontent.com... 151.101.88.133
Connecting to raw.githubusercontent.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

***** WARNING *****
This program will erase ALL THE DATA on the target USB flash drive and install SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER.
Before the installation, you must:
1. Back up all the data on the target USB flash drive.
2. Unplug all the USB flash drives except the target one.

***** 警告 *****
本程序将清除目标U盘中全部数据并安装SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER。
开始安装前，你必须：
1. 备份目标U盘上的全部数据。
2. 除目标U盘外，移除一切其它U盘。

Do you want to start the installation right now? (YES/NO): y
(install_online): 10001 ***** START INSTALLATION *****
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2019-11-12 00:38:44--  https://raw.githubusercontent.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/master/script_bootloader/bin/drive_modifier
Resolving raw.githubusercontent.com... 151.101.88.133
Connecting to raw.githubusercontent.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

(drive_modifier): 10133 ***** ERROR: CAN'T DETECT THE USB FLASH DRIVE *****
(install_online): 10001 ***** ABORT INSTALLATION *****

Keenni, 2019-11-11:
请问是什么原因？

JACK-THINK, 2019-11-15:
问题在于路由器没有正确挂载U盘

执行`cat /tmp/usb.log`，如果没有下述行，则U盘挂载失败
```
[usb_dbg: asus_sd] (sda): Success!
```

closed 2019-11-19 by JACK-THINK