JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #29: 请教新版v2ray的配置

huxuan, 2019-12-04:
首先感谢作者的付出，在AC86上试用了这套系统，感觉非常好，但是v2ray一直不工作感觉是配置的问题，尤其是希望实现的透明代理，还望能指点一下。已经尝试如下：

1. 安装没有问题，service以及monit的监控正常，非科学上网也是正常的
2. 参考了这里 [1] 的配置，能够在路由器内部翻墙，未进一步实现透明代理
3. 尝试用命令行直接执行启动命令

    `./usr/v2ray -config ./etc/config_proxy.json`

    在路由器内执行

    `curl google.com -x socks5://127.0.0.1:1080`

    返回错误是

    `curl: (52) Empty reply from server`

    v2ray也会提示错误

    ` v2ray.com/core/app/proxyman/outbound: failed to process outbound traffic > v2ray.com/core/proxy/vmess/outbound: connection ends > v2ray.com/core/proxy/vmess/outbound: failed to read header > v2ray.com/core/proxy/vmess/encoding: failed to read response header > EOF`

4. 确认过服务器时间没有问题
5. `config_proxy.json` 中只是参照文档[2]修改了对应条目，多修改了一个`alterID`因为我的服务器设置是`4`

[1] https://guide.v2fly.org/app/tproxy.html
[2] https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/script_bootloader/usr/v2ray/README_zh-CN.md

JACK-THINK, 2019-12-08:
@huxuan 
原来的说明文件已过时，[新版](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/script_bootloader/usr/v2ray/README_zh-CN.md)已出，请参阅

试用后请回复结果，谢谢

huxuan, 2019-12-08:
白天在外面浪了，遇到一个小问题，没有找到更新系统的方法，还是说只能重装[1]？

[1] https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_zh-CN.md#%E9%87%8D%E8%A3%85

JACK-THINK, 2019-12-09:
@huxuan 
昨天的更新改动内容幅度不大，无需重装整套系统，做如下变更即可：
1. 下载新安装包，解压`script_bootloader/usr/fwd/bin/fwd`，覆盖系统中`/opt/bin/fwd`和`/opt/script_bootloader/usr/fwd/bin/fwd`
2. 删除系统中现有的`/opt/script_bootloader/usr/shadowsocks`、`/opt/script_bootloader/usr/v2ray`、`/opt/script_bootloader/usr/overture`三个目录以及`/opt/script_bootloader/tmp/`目录下的全部文件。解压新安装包`script_bootloader/usr/shadowsocks`、`script_bootloader/usr/v2ray`、`script_bootloader/usr/overture`目录，上传至相应位置，执行`/opt/script_bootloader/bin/addons_install`重新安装插件。注意，安装前务必先修改各插件使用说明中提到的相应内容

huxuan, 2019-12-09:
应该没啥问题了，科学上网完美，谢谢~

closed 2019-12-09 by huxuan