JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #30: tuf-ax 3000 安裝時報錯 ，另外能不能集成AdGuard Home 插件？

fishredleung, 2019-12-17:
安裝失敗，顯示以下信息

(drive_modifier): 2965 ***** USB FLASH DRIVE IS READY *****
--2019-12-18 00:37:36--  https://api.github.com/repos/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/releases/latest
Resolving api.github.com... 140.82.113.5
Connecting to api.github.com|140.82.113.5|:443... connected.
OpenSSL: error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version
Unable to establish SSL connection.
http://: Invalid host name.
chmod: /tmp/mnt/ASUS_ROUTER/script_bootloader/: No such file or directory
/tmp/mnt/ASUS_ROUTER/sbl_install_log will be generated.
/tmp/install_online: line 206: /tmp/mnt/ASUS_ROUTER/script_bootloader/bin/gotee/gotee-linux-arm: not found
/tmp/install_online: line 206: /tmp/mnt/ASUS_ROUTER/script_bootloader/bin/addons_install: not found
/tmp/install_online: line 206: /opt/bin/tee: not found
***** STAGE 3: SUCCEEDED *****
mv: can't rename '/tmp/sbl_install_log': No such file or directory
(install_online): 2827 ***** INSTALLATION HAS BEEN COMPLETE *****
(install_online): 2827 ***** REBOOTING THE ROUTER *****

JACK-THINK, 2019-12-18:
网络故障，你的运营商污染了github。换个时间段安装，一般情况下早间OK。

AdGuard Home以后可能作为插件添加

JACK-THINK, 2019-12-20:
@fishredleung 
14.2&14.0版本已添加AdGuard Home（仅适用于arm64架构）

fishredleung, 2019-12-20:
與我無緣啊，ax3000 是armv7

JACK-THINK, 2019-12-21:
@fishredleung 
这个实在是没办法了。AdGuardHome官方的预编译版本AdGuardHome_linux_arm.tar.gz
不支持华硕路由器用的这种残缺的arm架构。已经有许多人在项目AdGuardHome中提了issue，等官方解决吧……

fishredleung, 2019-12-21:
昨晚又試了一下，還是顯示同樣的錯誤安裝不了，我已經把dns換成8.8.8.8 還是不行哦

fishredleung, 2019-12-21:
> @fishredleung
> 这个实在是没办法了。AdGuardHome官方的预编译版本AdGuardHome_linux_arm.tar.gz
> 不支持华硕路由器用的这种残缺的arm架构。已经有许多人在项目AdGuardHome中提了issue，等官方解决吧……

不是還有一個Linux 386版本麼？
能不能設一個指令自動根據cpu選擇

JACK-THINK, 2019-12-25:
@fishredleung 
1. 不知道你在什么地区，我这里只有早上5点-8点可以安装，晚上从未成功安装
2. 你的路由器是ARM架构，不可能安装AdGuardHome_linux_386.tar.gz版本，这是给x86架构的电脑用的
3. 我的代码当然是根据CPU类型自动选择合适的版本安装。但是TUF-AX3000这种采用被阉割了硬浮点的CPU的路由器无法运行AdGuardHome官方的ARM发行版。你可以自己编译源码或等待官方解决

fishredleung, 2019-12-25:
感謝你的解答.
我改天我早上試試能不能安裝吧！
官方版如果不能裝的話，我試試學著編譯吧，現在AdGuard home 在樹莓派上不太方便。哈哈

fishredleung, 2019-12-30:
1. 今早特地試了四次，每次都不行哦，是玄學嗎？
2. AdGuard home arm32bit 版本應該適合大部分路由器吧，可以加入嗎？因為樹莓派那些都能跑，路由器也行吧。

`admin@TUF-AX3000-DEC8:/tmp/home/root# cd /tmp && wget -q -O /tmp/install_onl
ine --no-check-certificate "https://raw.githubusercontent.com/JACK-THINK/SCR
IPTS-BOOTLOADER-FOR-ASUS-ROUTER/master/script_bootloader/bin/install_online"
 && chmod 777 /tmp/install_online && /tmp/install_online
***** WARNING *****
This program will erase ALL THE DATA on the target USB flash drive and install SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER.
Before the installation, you must:
1. Back up all the data on the target USB flash drive.
2. Unplug all the USB flash drives except the target one.

***** 警告 *****
本程序将清除目标U盘中全部数据并安装SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER。
开始安装前，你必须：
1. 备份目标U盘上的全部数据。
2. 除目标U盘外，移除一切其它U盘。

Do you want to start the installation right now? (YES/NO): yes
(install_online): 2792 ***** START INSTALLATION *****
--2019-12-31 06:53:02--  https://raw.githubusercontent.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/master/script_bootloader/bin/drive_modifier
Resolving raw.githubusercontent.com... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com|151.101.0.133|:443... connected.
WARNING: cannot verify raw.githubusercontent.com's certificate, issued by '/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 High Assurance Server CA':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 200 OK
Length: 2902 (2.8K) [text/plain]
Saving to: '/tmp/drive_modifier'

/tmp/drive_modifie 100%[==================>]   2.83K  --.-KB/s   in 0s     

2019-12-31 06:53:02 (5.91 MB/s) - '/tmp/drive_modifier' saved [2902/2902]

(drive_modifier): 3385 ***** RE-PARTITION THE USB FLASH DRIVE *****

Command (m for help): 
Disk /dev/sda: 8000 MB, 8000110592 bytes
247 heads, 62 sectors/track, 1020 cylinders
Units = cylinders of 15314 * 512 = 7840768 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1               1        1020     7810109  83 Linux

Command (m for help): Building a new DOS disklabel. Changes will remain in memory only,
until you decide to write them. After that the previous content
won't be recoverable.

Command (m for help): Command action
   e   extended
   p   primary partition (1-4)
Partition number (1-4): First cylinder (1-1020, default 1): Using default value 1
Last cylinder or +size or +sizeM or +sizeK (1-1020, default 1020): Using default value 1020

Command (m for help): 
Disk /dev/sda: 8000 MB, 8000110592 bytes
247 heads, 62 sectors/track, 1020 cylinders
Units = cylinders of 15314 * 512 = 7840768 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1               1        1020     7810109  83 Linux

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table
(drive_modifier): 3385 ***** LABEL OF THE USB FLASH DRIVE IS ASUS_ROUTER *****
mke2fs 1.42.13 (17-May-2015)
/dev/sda1 contains a ext3 file system labelled 'ASUS_ROUTER'
        last mounted on /tmp/mnt/ASUS_ROUTER(1) on Tue Dec 31 06:53:19 2019
Creating filesystem with 1952527 4k blocks and 488640 inodes
Filesystem UUID: b153131a-62df-45e9-bdb2-35396c93f1d9
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done 

(drive_modifier): 3385 ***** USB FLASH DRIVE IS READY *****
--2019-12-31 06:58:29--  https://api.github.com/repos/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/releases/latest
Resolving api.github.com... 140.82.114.5
Connecting to api.github.com|140.82.114.5|:443... connected.
OpenSSL: error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version
Unable to establish SSL connection.
(install_online): 2792 ***** CONNECTION FAILED *****
(install_online): 2792 ***** ABORT INSTALLATION *****
admin@TUF-AX3000-DEC8:/tmp#  `

JACK-THINK, 2019-12-31:
1. 看来你的网络环境在“墙内”，Github的污染问题难以解决。彻底的解决方法就是把这个路由器连接到一个已经可以“科学上网”的网关后面做客户端，然后执行安装程序即可。但是这是一个“鸡生蛋，蛋生鸡”的问题……绝大多数情况下，你只能碰运气。你也可以试试[离线安装方法](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_zh-CN.md#离线安装)，昨天有小伙伴成功了。

2. 目前AdGuardHome官方提供的预编译包`AdGuardHome_linux_arm.tar.gz`是无法在armv7架构的华硕路由器上运行的。已经有很多人给AdGuardHome提交了issue，也许下一版发布时官方会提供armv5的预编译包（华硕路由器armv7架构CPU阉割了硬浮点，导致只能运行armv5这种只有软浮点的预编译程序）。若要解决此问题，必须按下述操作步骤重新编译源码：

   1. 获取AdGuardHome源码
      ```
      git clone https://github.com/AdguardTeam/AdGuardHome
      cd AdGuardHome
      vim release.sh
      ```
   2. 修改如下行
      `CHANNEL=$channel GOOS=linux GOARCH=arm GOARM=6 f`
      为
      `CHANNEL=$channel GOOS=linux GOARCH=arm GOARM=5 f`
   
   3. 执行 `./release.sh`
   
   4. `./dist/AdGuardHome_linux_arm.tar.gz` 就是你需要的程序

fishredleung, 2019-12-31:
> @fishredleung
> 
> 1. 看来你的网络环境在“墙内”，Github的污染问题难以解决。彻底的解决方法就是把这个路由器连接到一个已经可以“科学上网”的网关后面做客户端，然后执行安装程序即可。但是这是一个“鸡生蛋，蛋生鸡”的问题……绝大多数情况下，你只能碰运气。你也可以试试[离线安装方法](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_zh-CN.md#%E7%A6%BB%E7%BA%BF%E5%AE%89%E8%A3%85)，昨天有小伙伴成功了。
> 2. 目前AdGuardHome官方提供的预编译包`AdGuardHome_linux_arm.tar.gz`是无法在armv7架构的华硕路由器上运行的。已经有很多人给AdGuardHome提交了issue，我也给AdGuardHome提交了针对此问题的[解决方法](https://github.com/AdguardTeam/AdGuardHome/issues/1239)。也许下一版发布时官方会提供armv5的预编译包（华硕路由器armv7架构CPU阉割了硬浮点，导致只能运行armv5这种只有软浮点的预编译程序；树莓派用的armv7架构CPU支持硬浮点）。若要解决此问题，必须按下述操作步骤重新编译源码：
>    
>    1. 获取AdGuardHome源码
>       ```
>       git clone https://github.com/AdguardTeam/AdGuardHome
>       cd AdGuardHome
>       vim release.sh
>       ```
>    2. 修改如下行
>       `CHANNEL=$channel GOOS=linux GOARCH=arm GOARM=6 f`
>       为
>       `CHANNEL=$channel GOOS=linux GOARCH=arm GOARM=5 f`
>    3. 执行 `./release.sh`
>    4. `./dist/AdGuardHome_linux_arm.tar.gz` 就是你需要的程序

很詳細，謝謝！

後天再調個鬧鐘起床折騰。
我沒在牆內。

JACK-THINK, 2020-01-02:
1. Adguard官方已经把我的方案并入版本v0.101，等新版发布吧

2. 没在墙内？难道你那边也有DNS污染？？我的日本和美国测试节点从没出现过连接失败的问题。

fishredleung, 2020-02-09:
`(drive_modifier): 3467 SUCCESS: SanDisk Cruzer Blade 1.00 IS READY
--2020-02-09 13:47:10--  https://api.github.com/repos/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/releases/latest
Resolving api.github.com... 140.82.113.5
Connecting to api.github.com|140.82.113.5|:443... connected.
OpenSSL: error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version
Unable to establish SSL connection.
(install_online): 3227 NOTICE: CONNECTION FAILED
(install_online): 3227 FAILURE: ONLINE-INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER
`
還是不行，感覺我要放棄了

guanzi008, 2020-02-10:
> @fishredleung
> 
> 1. 不知道你在什么地区，我这里只有早上5点-8点可以安装，晚上从未成功安装
> 2. 你的路由器是ARM​​架构，不可能安装AdGuardHome_linux_386.tar.gz版本，这是给x86架构的电脑用的
> 3. 但是，TUF-AX3000这种采用阉割了硬浮点的CPU的路由器无法运行AdGuardHome官方的ARM发行版。您可以自己编译源码或等待官方解决
很好编译啊

JACK-THINK, 2020-02-11:
> `(drive_modifier): 3467 SUCCESS: SanDisk Cruzer Blade 1.00 IS READY --2020-02-09 13:47:10-- https://api.github.com/repos/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/releases/latest Resolving [...]

阅读手册，使用离线安装方法可解决此问题。

guanzi008, 2020-02-11:
wget --no-check-certificate 地址
禁用ssl
curl -k

guanzi008, 2020-02-11:
> > `(drive_modifier): 3467 SUCCESS: SanDisk Cruzer Blade 1.00 IS READY --2020-02-09 13:47:10-- https://api.github.com/repos/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/releases/latest Resolving api.github.com... 140.82.113.5 Connecting to api.github.com|140.82.113.5|:443... connected. OpenSSL: error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version Unable to establish SSL connection. (install_online): 3227 NOTICE: CONNECTION FAILED (install_online): 3227 FAILURE: ONLINE-INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER `
> > 還是不行，感覺我要放棄了
> 
> 更新至最新版固件。wget版本必须高于1.20才能解决SSL错误

https://blog.csdn.net/lilongsy/article/details/97630027

JACK-THINK, 2020-02-11:
> https://blog.csdn.net/lilongsy/article/details/97630027

@guanzi008 

此方法不行。看[install_online源码](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/script_bootloader/bin/install_online)的第264行，已经有`--no-check-certificate`了。此处报错是因为华硕路由器固件内的wget（版本低于1.20）有问题，只能通过刷最新固件把wget升级至1.20才能解决。

edwinhuish, 2020-02-12:
格式化好U盘，手动下载zip上传到路由器后解压，运行 ```/script_bootloader/bin/``` 下的 install 手动安装吧。

fishredleung, 2020-02-12:
> 格式化好U盘，手动下载zip上传到路由器后解压，运行 `/script_bootloader/bin/` 下的 install 手动安装吧。

這樣做的話只能治標吧，以後更新切要離線安裝嗎。。。

fishredleung, 2020-02-12:
> @guanzi008 [...]

Firmware 已經是最新了，可以單獨更新wget插件？

JACK-THINK, 2020-02-13:
> Firmware 已經是最新了，可以單獨更新wget插件？

@fishredleung 

执行`/usr/sbin/wget -V`，看看固件中wget的版本号。如果小于1.20，那么只能使用离线安装方法。

edwinhuish, 2020-02-13:
> > 格式化好U盘，手动下载zip上传到路由器后解压，运行 `/script_bootloader/bin/` 下的 install 手动安装吧。
> 
> 這樣做的話只能治標吧，以後更新切要離線安裝嗎。。。

我没有考虑过更新，一般路由器折腾好之后，基本就是扔在角落里吃灰，不会去动它。

偶尔有问题，自己修复一下就好了。

closed 2020-03-02 by JACK-THINK

fishredleung, 2020-03-17:
請問新版本可以識別armv7 cpu了嗎，AdGuard home官方已經更新

JACK-THINK, 2020-03-18:
@fishredleung 
从16.1&16.1开始支持全平台Adguard Home

melxman, 2020-03-19:
首先先感谢下作者，16.1确实可以在66u-b1上用adhome了。但是下面我有了两个问题。
1:可能是固件兼容性？(我用的是ks的改版梅林)
开启梯子，ad就不再解析和过滤，关上就好(关梯子过滤解析都没问题)
梯子里关于dns的部分我实验了各种组合。但是一直不成功。
不知道作者有没有好的解决办法？
2:Monit界面(35490端口)，一直重复提示登录。但是adhome一次就登录成功。我确认帐号密码没有错误。不知道问题在哪儿。且已经重启过路由测试

fishredleung, 2020-03-22:
請問一下，路由器的53端口被佔用了，AdGuard home提示端口為5353，但路由器dns欄不支持填寫port number ,有甚麼方法才可以令流量被AdGuard home過慮？

JACK-THINK, 2020-03-22:
> 首先先感谢下作者，16.1确实可以在66u-b1上用adhome了。但是下面我有了两个问题。
> 1:可能是固件兼容性？(我用的是ks的改版梅林)
> 开启梯子，ad就不再解析和过滤，关上就好(关梯子过滤解析都没问题)
> 梯子里关于dns的部分我实验了各种组合。但是一直不成功。
> 不知道作者有没有好的解决办法？
> 2:Monit界面(35490端口)，一直重复提示登录。但是adhome一次就登录成功。我确认帐号密码没有错误。不知道问题在哪儿。且已经重启过路由测试

@melxman 
暂不能同时使用。因为梯子和Adguardhome都要做dnsmasq的唯一上游服务器。虽然有改进方法，但是非常麻烦，而且兼容性无法保证，只能先这样了

JACK-THINK, 2020-03-22:
@fishredleung 
正确安装并启用后，adguardhome应提示监听端口为2153，不是5353

fishredleung, 2020-03-22:
> @fishredleung
> 正确安装并启用后，adguardhome应提示监听端口为2153，不是5353

在DNS 不能輸入port 的程況下，路由器是不能以AdGuard home 作全局過濾嗎？
有辦法在AdGuard 開啟後，所有路由器上的數據都以AdGuard home 作為dns嗎？

melxman, 2020-03-23:
> @melxman
> 暂不能同时使用。因为梯子和Adguardhome都要做dnsmasq的唯一上游服务器。虽然有改进方法，但是非常麻烦，而且兼容性无法保证，只能先这样了

十分感谢回答