JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #32: 怎么离线安装,教程只有在线

magian1127, 2019-12-28:
国内路由器访问不了 raw.githubusercontent.com
然后自己折腾了半天 离线也没成功..
路由器是 ax88u 官方最新固件

JACK-THINK, 2019-12-29:
@magian1127 
因为有很多插件的安装也要访问GITHUB，所以就干掉了离线安装教程。也许近期我会再写一份离线安装教程。建议你早晨7-9点进行在线安装，一般都能成功。晚上基本不行

magian1127, 2019-12-29:
@JACK-THINK 
果然早上可以...但是还是折腾到了中午.. 就访问不了了..
主要是drive_modifier 过不了..

`
(drive_modifier): 3736 ***** RE-PARTITION THE USB FLASH DRIVE *****

Command (m for help): 
Disk /dev/sda: 7759 MB, 7759462400 bytes
239 heads, 62 sectors/track, 1022 cylinders
Units = cylinders of 14818 * 512 = 7586816 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1               1        1022     7571967  83 Linux

Command (m for help): Building a new DOS disklabel. Changes will remain in memory only,
until you decide to write them. After that the previous content
won't be recoverable.

Command (m for help): Command action
   e   extended
   p   primary partition (1-4)
Partition number (1-4): First cylinder (1-1022, default 1): Using default value 1
Last cylinder or +size or +sizeM or +sizeK (1-1022, default 1022): Using default value 1022

Command (m for help): 
Disk /dev/sda: 7759 MB, 7759462400 bytes
239 heads, 62 sectors/track, 1022 cylinders
Units = cylinders of 14818 * 512 = 7586816 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1               1        1022     7571967  83 Linux

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table
(drive_modifier): 3736 ***** ERROR: CAN'T RE-PARTITION THE USB FLASH DRIVE *****
`

JACK-THINK, 2019-12-29:
@magian1127 离线安装文档已出，你先试试。如果drive_modifier还是过不了，你用离线安装，到格式化U盘时，自己用`fdisk`命令手动格式化U盘并把它挂载到`/tmp/mnt/ASUS_ROUTER`，后续步骤不变。如果还不行，找个文本编辑器直接打开drive_modifier逐行测试即可。

magian1127, 2019-12-29:
搞定了..谢谢,我手动格式化硬盘了..然后离线安装成功了

closed 2019-12-29 by magian1127