JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #34: 如何设置密码？

seamusic, 2020-02-03:
您好。在安装过程中，由于安装的时间太久了，在最后的用户名和密码时断开了。所以现在不知道如何登录？有解决办法吗？

(addons_install): 5476 INFO: USERNAME: login username of the router
(addons_install): 5476 INFO: INPUT THE VALUE: (addons_install): 5476 INFO: INPUT THE VALUE AGAIN TO CONFIRM:

这一步因为没有及时操作，已经断开了。

seamusic, 2020-02-03:
已使用默认密码admin/monit成功登录。
不知道这个中断会不会有其它问题？

seamusic, 2020-02-03:
貌似啥软件都没默认安装。

seamusic, 2020-02-03:
后续步骤可以参考 https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_zh-CN.md#%E7%A6%BB%E7%BA%BF%E5%AE%89%E8%A3%85

JACK-THINK, 2020-02-03:
@seamusic 
> 已使用默认密码admin/monit成功登录。
> 不知道这个中断会不会有其它问题？

会有其他问题。此情况下，一定要执行
```
/tmp/mnt/ASUS_ROUTER/script_bootloader/bin/addons_install
```
重新设置密码。否则会有很多插件无法正常工作。

seamusic, 2020-02-03:
嗯，谢谢。还有个问题，在安装ss的时候，因为网络的问题安装不完全。如何重新安装？有没有卸载的插件的脚本？

seamusic, 2020-02-03:
找到了，可以直接到各个插件下有对应的脚本。直接执行/opt/script_bootloader/usr/overture/bin/overture_install安装就可以了。

JACK-THINK, 2020-02-03:
@seamusic 
有
```
/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/shadowsocks/bin/shadowsocks_uninstall
```

closed 2020-02-03 by JACK-THINK