JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #37: sed: invalid option -- E

post707, 2020-03-05:
admini@RT-AC68U:/tmp# /tmp/prerequisite_checker
sed: invalid option -- E
BusyBox v1.17.4 (2019-04-09 12:05:08 CST) multi-call binary.

Usage: sed [-efinr] SED_CMD [FILE]...

Options:
        -e CMD  Add CMD to sed commands to be executed
        -f FILE Add FILE contents to sed commands to be executed
        -i      Edit files in-place (else sends result to stdout)
        -n      Suppress automatic printing of pattern space
        -r      Use extended regex syntax

If no -e or -f, the first non-option argument is the sed command string.
Remaining arguments are input files (stdin if none).

extended regular expression: NOT WORKING
(prerequisite_checker): 1767 NOTICE: MISSING ESSENTIAL FEATURES

JACK-THINK, 2020-03-06:
Report version of firmware.

JACK-THINK, 2020-03-08:
@post707 
Report version of firmware.

chfulinliu, 2020-03-24:
same problem for RT-ACRH17 with official firmware: 3.0.0.4.382_51569

chfulinliu, 2020-03-24:
After replacing all "sed -E" with "sed -r" in all scripts, the problem can be solved. @post707

JACK-THINK, 2020-03-29:
> After replacing all "sed -E" with "sed -r" in all scripts, the problem can be solved. @post707

@chfulinliu 

You've got the point! Standard `sed` needs to support both option `-E` and option `-r`, which have the same function. But some variants of `sed` just support one of them, leaving a lot of trouble in compatibility. If you decide to change option `-E` to `-r`, You REALLY need to be careful enough to make sure all the option `-E` replaced without an error. Otherwise **CATASTROPHE** would come up.

closed 2020-03-29 by JACK-THINK

lisbethw1130, 2020-10-28:
Hi, I'm using RT-AC55UHP with version 3.0.0.4.382_52236-ga0f880d, also get this error

Is that mean I have to download offline version and mass replace sed -E with sed -r, right?