JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #41: GitHub连不上的原因？

jiujiagou, 2020-04-04:
/tmp/sbl_download_l 100%[===================>]   2.04K  --.-KB/s    in 0s      

2020-04-04 16:25:07 (7.84 MB/s) - '/tmp/sbl_download_link.html' saved [2087/2087]

Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2020-04-04 16:25:07--  https://api.github.com/repos/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/tarball/16.1&16.1
Resolving api.github.com... 13.250.168.23
Connecting to api.github.com|13.250.168.23|:443... failed: Connection refused.
(install_online): 6464 NOTICE: CONNECTION FAILED
(install_online): 6464 FAILURE: ONLINE-INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER

安装中如上报错。请教一下大家怎么解决的呢？

edwinhuish, 2020-04-04:
在国内，遇到这种情况是正常的，我之前建议过作者放 gitee，但由于需要实名验证，他并没有接受建议。。

SO.... 唯一解决方法就是离线安装。

jiujiagou, 2020-04-04:
> 在国内，遇到这种情况是正常的，我之前建议过作者放 gitee，但由于需要实名验证，他并没有接受建议。。
> 
> SO.... 唯一解决方法就是离线安装。

感谢，我尝试下离线

closed 2020-04-07 by JACK-THINK