JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #43: 安装插件时如果获取自GitHub，有时候会获取不到版本号

leen2175, 2020-04-06:
日志如下：
`(v2ray_install): 6225 *--------- INSTALL V2RAY ON RT-AX58U_EPA ---------*
(v2ray_install): 6225 INFO: ENTWARE_ENABLE SERVICE HAS BEEN LOADED
(v2ray_install): 6225 INFO: MONIT MANAGEMENT SYSTEM HAS BEEN DEPLOYED
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2020-04-06 17:20:35--  https://github.com/v2ray/v2ray-core/releases/download//v2ray-linux-arm.zip
Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-04-06 17:20:35 ERROR 404: Not Found.

(v2ray_install): 6225 NOTICE: CONNECTION FAILED
(v2ray_install): 6225 FAILURE: INSTALL V2RAY`

leen2175, 2020-04-06:
有时候获取到版本号也不能安装成功
`(ariang_install): 8669 *--------- INSTALL ARIANG ON RT-AX58U_EPA ---------*
(ariang_install): 8669 INFO: ENTWARE_ENABLE SERVICE HAS BEEN LOADED
(ariang_install): 8669 INFO: MONIT MANAGEMENT SYSTEM HAS BEEN DEPLOYED
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2020-04-06 17:22:07--  https://github.com/mayswind/AriaNg/releases/download/1.1.4/AriaNg-1.1.4-AllInOne.zip
Resolving github.com... 13.250.177.223
Connecting to github.com|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/58753834/9f0e2900-ea1c-11e9-9432-447d200db0ee?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200406%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200406T092022Z&X-Amz-Expires=300&X-Amz-Signature=4c9c927808df847db784ade3380248f969abdd9f6920e3e3eaef4a83391957a4&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAriaNg-1.1.4-AllInOne.zip&response-content-type=application%2Foctet-stream [following]
--2020-04-06 17:22:07--  https://github-production-release-asset-2e65be.s3.amazonaws.com/58753834/9f0e2900-ea1c-11e9-9432-447d200db0ee?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200406%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200406T092022Z&X-Amz-Expires=300&X-Amz-Signature=4c9c927808df847db784ade3380248f969abdd9f6920e3e3eaef4a83391957a4&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAriaNg-1.1.4-AllInOne.zip&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com... 52.217.40.100
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com|52.217.40.100|:443... connected.
OpenSSL: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
Unable to establish SSL connection.
(ariang_install): 8669 NOTICE: CONNECTION FAILED
(ariang_install): 8669 FAILURE: INSTALL ARIANG`

JACK-THINK, 2020-04-07:
@leen2175 

我自己也受此问题影响……

近来半个月XXX对Github的污染愈发严重了……这问题不是我等小民能解决的

closed 2020-04-07 by JACK-THINK

befantasy, 2020-06-08:
遇到同样问题。
看了一下v2ray addon的安装脚本
/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/v2ray/bin/v2ray_install

脚本中的下载连接为

` DOWNLOAD_LINK="https://github.com/v2ray/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`

执行脚本看到的报错

>  https://github.com/v2ray/v2ray-core/releases/download//v2ray-linux-arm.zip
> Resolving github.com... 13.229.188.59
> Connecting to github.com|13.229.188.59|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2020-06-08 14:15:21 ERROR 404: Not Found.
> 

可以看到url中${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE的变量都正常解析出来了，而${LATEST_VERSION}变量没有正常解析。导致404 not found。

继续查看安装脚本，${LATEST_VERSION}变量

` LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`

单独执行

> /opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" 

返回完全正常。正常返回了v2ray的所有latest release信息。

至此，基本判定作者的sed正则表达式写的有问题，或者是v2ray的release命名方式改了，导致无法解析latest version信息，安装失败。不过我对正则表达式一窍不通，所以也无法提交PR。

搜了好几个issues，都有类似的问题，不但v2ray无法安装，我试了好几个addon都无法安装。我觉得作者的回答十分不负责任，全都看都不看，一概推给github被污染。。。

结合项目的安装过程中，U盘分区，entware安装等过程出过好多莫名其妙的问题。总体感觉这个项目很不靠谱，本来寄予厚望，结果失望至极。