JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #45: unbound also is please

hanston, 2020-04-07:
extremely grateful if you could try unbound in your scropt
unbound with adguard home will remove raspberry pie with pie hole from the network
sorry if this was a wrong post

JACK-THINK, 2020-04-08:
@hanston 

You're welcome. Could you describe your aim more clearly? As far as I know, unbound is a DNS server, which could be replaced by the built-in dnsmasq easily. Is there anything that could be done with unbound exclusively? Please be more specific.

hanston, 2020-04-08:
what i am using now is unbound and pie hole on a raspberry pie 
pages open faster and it is a more secure dns 
if some how unbound and adguard home on a asus router it will remove the need to use raspberry pie in between router and the internet
as long as unbound is always ON and building a cache the pages load faster
unbound is already available for asus router here 
https://www.snbforums.com/threads/release-unbound_manager-manager-installer-utility-for-unbound-recursive-dns-server.61669/
if this can b integrated in your script

jumpsmm7, 2020-05-16:
> 
> 
> @hanston
> 
> You're welcome. Could you describe your aim more clearly? As far as I know, unbound is a DNS server, which could be replaced by the built-in dnsmasq easily. Is there anything that could be done with unbound exclusively? Please be more specific.

Unbound is a authoritative recursive caching dns server. With unbound, you are your own DNS server as you request dns lookups directly from the root servers. where as DNSMasq is used primarily as a forwarder and for DHCP capabilities.