JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #47: 小白请问怎么离线安装插件？？？

bendfeng, 2020-04-12:
小白请问怎么离线安装插件？？？

bendfeng, 2020-04-12:
admin@RT-AC86U-7EE0:/tmp/home/root/Downloads# /tmp/mnt/ASUS_ROUTER/script_bootloader/usr/v2ray/bin/v2ray_install
(v2ray_install): 6394 *--------- INSTALL V2RAY ON RT-AC86U ---------*
(v2ray_install): 6394 INFO: ENTWARE_ENABLE SERVICE HAS BEEN LOADED
(v2ray_install): 6394 INFO: MONIT MANAGEMENT SYSTEM HAS BEEN DEPLOYED
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2020-04-12 11:41:34--  https://github.com/v2ray/v2ray-core/releases/download//v2ray-linux-arm64.zip
Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-04-12 11:41:35 ERROR 404: Not Found.

(v2ray_install): 6394 NOTICE: CONNECTION FAILED
(v2ray_install): 6394 FAILURE: INSTALL V2RAY

JACK-THINK, 2020-04-14:
@bendfeng 
当前插件无法离线安装