JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #49: v2ray插件的配置疑问

Heseli, 2020-04-24:
你好：
      我按照你的说明文档安装好了SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER并安装了v2ray插件。根据v2ray插件的说明，启动界面如下：

[image]
但完全无法上网，包括www.baidu.com也无法访问，必须关闭V2RAY_TRANSPARENT_PROXY才能正常上网，请问是什么问题呢？
      我已经按照v2ray插件的说明在/opt/script_bootloader/usr/v2ray/bin/fw_v2ray_transparent_proxy_enable.service文件中修改了参数TPROXY_MODULES_STATUS=false，请问还有哪个地方配置不对？v2ray配置文件我更改了outbounds中的vmess信息，inbounds不更改，这样正确么？
    期待你的回复，谢谢！

JACK-THINK, 2020-04-29:
@Heseli 

对端的v2ray服务器配置正确吗？

Heseli, 2020-04-29:
服务器端的v2ray应该是正常的，我用手机和电脑客户端都能正常连上使用。在路由器端的v2ray，我就是更改了/opt/ASUS_ROUTER/script_bootloader/usr/v2ray/etc下的config_proxy.json，配置的各个参数应该没有错，就不知道还有哪个地方出错了，我附上我的config_proxy.json，请帮忙查看一下，谢谢！

{
    "inbounds":
    [
        {
            "port":1080,
            "listen":"0.0.0.0",
            "protocol":"socks",
            "settings":
            {
                "auth":"noauth",
                "udp":true,
                "ip":"127.0.0.1",
                "userLevel":1
            },
            "tag":"LAN_inbound_socks_VPS"
        },
        {
            "port":2053,
            "listen":"0.0.0.0",
            "protocol":"dokodemo-door",
            "settings":
            {
                "address":"223.5.5.5",
                "port":53,
                "network":"tcp,udp",
                "timeout":0,
                "followRedirect":false,
                "userLevel":1
            },
            "tag":"DNS_local_dokodemo_VPS"
        },
        {
            "port":45154,
            "listen":"0.0.0.0",
            "protocol":"dokodemo-door",
            "settings":
            {
                "network":"tcp,udp",
                "timeout":30,
                "followRedirect":true,
                "userLevel":1
            },
            "streamSettings":
            {
                "sockopt":
                {
                    "tproxy":"redirect"
                }
            },
            "tag":"LAN_local_dokodemo_VPS"
        },
        {
            "port":22815
            "listen":"",
            "protocol":"vmess",
            "settings":
            {
                "clients":
                [
                    {
                        "id":"",
                        "level":1,
                        "alterId":64
                    }
                ],
                "disableInsecureEncryption":true
            },
            "streamSettings":
[... 358 lines cut ...]
                "type":"field",
                "inboundTag":
                [
                    "WAN_inbound_shadowsocks_Router"
                ],
                "outboundTag":"Router_outbound"
            }
        ]
    },
    "policy":
    {
        "levels":
        {
            "1":
            {
                "handshake":4,
                "connIdle":300,
                "uplinkOnly":2,
                "downlinkOnly":5,
                "statsUserUplink":false,
                "statsUserDownlink":false
            }
        }
    }
}

JACK-THINK, 2020-05-03:
@Heseli 
这个格式实在是一言难尽啊……
你直接运行`/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/v2ray/bin/v2ray_configure`，按照屏幕提示再重新配置一遍试试吧

Heseli, 2020-05-05:
Hi:
    我重新运行了v2ray_configure,仍然无法正常上网（只要启动了V2RAY_TRANSPARENT_PROXY就无法上网，连百度也上不了，只能关闭）。我的v2ray配置一再确认过应该没有问题。怀疑是不是更改/opt/script_bootloader/usr/v2ray/bin/fw_v2ray_transparent_proxy_enable.service 第86行为TPROXY_MODULES_STATUS=false这一项有错误？我提供一下目前我的路由器上这个文件目前的截图，请帮我确认一下是否更改正确，谢谢！

[image]

JACK-THINK, 2020-05-05:
@Heseli 

果然这里有问题。把现有“/opt/script_bootloader/usr/v2ray/bin/fw_v2ray_transparent_proxy_enable.service”的第86行改为`########## CHECK TPROXY MODULES ##########`，把第90行中`true`改为`false`，其它都不要动。试试这样改能否解决问题

Heseli, 2020-05-08:
很遗憾，更改了之后还是无法上网，唉。

closed 2020-05-11 by JACK-THINK