JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #54: (v2ray_install): 7183 FAILURE: INSTALL V2RAY

iPatrickW, 2020-04-25:
`Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-04-26 07:56:39 ERROR 404: Not Found.

(v2ray_install): 7183 NOTICE: CONNECTION FAILED
(v2ray_install): 7183 FAILURE: INSTALL V2RAY
`

JACK-THINK, 2020-04-26:
Github被ISP污染

closed 2020-04-26 by JACK-THINK

befantasy, 2020-06-08:
遇到同样问题。
看了一下v2ray addon的安装脚本
/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/v2ray/bin/v2ray_install

脚本中的下载连接为

` DOWNLOAD_LINK="https://github.com/v2ray/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`

执行脚本看到的报错

>  https://github.com/v2ray/v2ray-core/releases/download//v2ray-linux-arm.zip
> Resolving github.com... 13.229.188.59
> Connecting to github.com|13.229.188.59|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2020-06-08 14:15:21 ERROR 404: Not Found.
> 

可以看到url中${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE的变量都正常解析出来了，而${LATEST_VERSION}变量没有正常解析。导致404 not found。

继续查看安装脚本，${LATEST_VERSION}变量

` LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`

单独执行

> /opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" 

返回完全正常。正常返回了v2ray的所有latest release信息。

至此，基本判定作者的sed正则表达式写的有问题，或者是v2ray的release命名方式改了，导致无法解析latest version信息，安装失败。不过我对正则表达式一窍不通，所以也无法提交PR。

搜了好几个issues，都有类似的问题，不但v2ray无法安装，我试了好几个addon都无法安装。我觉得作者的回答十分不负责任，全都看都不看，一概推给github被污染。。。

结合项目的安装过程中，U盘分区，entware安装等过程出过好多莫名其妙的问题。总体感觉这个项目很不靠谱，本来寄予厚望，结果失望至极。