JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #57: Documentation & Queries

PC-Pilot, 2020-04-27:
Addons V17.0

WOL Addon - Whilst updating from Addons V15.0 to V17.0 (V17.0-17.0) I noticed that the user configurable script had been revised (updated) with a number of parameters changed.

I referred to "https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/wiki/Indexes-for-instructions (instructions for addons)" for an updated guide(s) however none of these links are currently working. Is there an alternate link to these (revised) documents?
-----------------------------------------------------------------------
General Query - Is there an simple way to apply the incremental updates as/when they are released? Or is it necessary to install (either online or offline) the entire release package? 

Additionally, would it be possible to provide a list (detailed alongside their release date/time) of the update changes, as these are always replaced on the release page by each subsequent package release as it is listed?  This would help users to assess the urgency of updating to a given release especially if the component(s) concerned have not been selected by the user in the first place and therefore do not represent a priority for upgrade.

JACK-THINK, 2020-04-28:
@PC-Pilot 

Thanks for your suggestion.

1. [This repo](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER-ADDONS) is deprecated. But when you need some addon which is removed from [here](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/tree/master/script_bootloader/usr), you can download it.

2. Read [How_to_Use](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_en-US.md), and you can get some information you need about update and upgrade.

3. To some extent, incremental update function has been accomplished. But, there might be some or a lot of  modifications which is incompatible with the former version. So, if a **MAJOR** version is published, like from `SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER_v16.X&XX.X` to `SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER_v17.X&XX.X`, users need to install it from scratch or execute upgrade **between 3rd to 20th every month**.

4. English documents haven't been finished except [How_to_Use](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_en-US.md). I'll start this work soon.

JackMerlin, 2020-05-15:
@JACK-THINK  
> English documents haven't been finished except [How_to_Use](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/How_to_Use_en-US.md). I'll start this work soon.

It may be a good idea to let anyone edit your github Wiki.