JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #59: Compatibility with Skynet

PC-Pilot, 2020-05-13:
Is "SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER" compatible with Skynet (https://github.com/Adamm00/IPSet_ASUS)??

If yes, can it be incorporated as an integral Addon??

PC-Pilot, 2020-05-14:
Can I ask the same for Diversion (https://diversion.ch/)...  and whether that can also be incorporated as an integral optional Addon ….as a free alternative to the existing adguardhome addon??

NB. I have experimented with the Skynet command line automated install (/usr/sbin/curl -s "https://raw.githubusercontent.com/Adamm00/IPSet_ASUS/master/firewall.sh" -o "/jffs/scripts/firewall" && chmod 755 /jffs/scripts/firewall && sh /jffs/scripts/firewall install) and found that it damaged either the SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER install and/or file structure. Monit was also inaccessible and WOL ceased to function...  likely my other addons were similarly disabled.  Recovery was only possible by removing Skynet and completely reinstalling and reconfiguring the SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER from scratch.

As my knowledge is rather lacking in these matters perhaps you could suggest a work-around install procedure or alternately ….. :) as an optional addon (alongside Diversion :) :)…..please) via Stage III of your excellent SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER automated installer.

Finally, in the hope that it is of some use for you to incorporate (either within Stage I, II or as an option at Stage III) the following script was discovered on GitHub and applied (without issue): See (https://github.com/Adamm00/ChannelHog).

JACK-THINK, 2020-05-15:
@PC-Pilot I'll dig into `diversion` in a few days to see if it can be integrated into my project.

JACK-THINK, 2020-05-15:
@PC-Pilot Project `Skynet` is too complicated for me. It can be a very hard job to detach it from ASUS-MERLIN if it depends on some unique features of ASUS-MERLIN

PC-Pilot, 2020-05-15:
Hi Jack and thank you again for responding so swiftly, very much appreciated!!

As a complete novice to scripting in general and the intricacies of the Asus Router and Merlin Firmware in particular I would like to express my appreciation for your innovative SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER which has enabled me to both access advanced features and also begin to comprehend just a little more about scripting, so thank you for that!

Naturally I can therefore understand why the complexities of Skynet and it's interdependence with the core Merlin firmware precludes its addition to your project and as an alternative would instead like to seek your advice as to how I might install it separately without damaging the file structure created by your SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER and thus access to the addons within.

For example would Skynet be better (in my case) installed manually and if so, which command line script(s) would be necessary to ensure compatibility with your SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER and the prerequisite install items such as Monit and Entware (which both seemingly share).

As a novice, I would greatly appreciate a step by step guide much like the one you prepared with images to explain your  SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER "offline" install process. Also, your advice as to the most appropriate install sequence (i.e. SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER and then Skynet .....or else with both reversed) to ensure compatibility.

If it is easier for you, the same form of guidance in respect of Diversion would of course also be most welcome...

I greatly appreciate your informative replies and very much value the SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER project which has encouraged me to explore other useful Addons.  Currently I am trying to understand the highly regarded YazFi (https://github.com/jackyaz/YazFi) - Guest Wi-Fi Network which seemingly enables simultaneous use of two VPN Clients... currently trying to understand the configuration!  Maybe something else worthy of incorporation in your project at some future point??

Very best regards,

PC Pilot

JackMerlin, 2020-05-15:
@JACK-THINK 
> Project `Skynet` is too complicated for me. It can be a very hard job to detach it from ASUS-MERLIN if it depends on some unique features of ASUS-MERLIN

`skynet` uses more iptables and ipset commands. so I think if don't change these two commands `skynet` should work well.

> I'll dig into `diversion` in a few days to see if it can be integrated into my project.

But it might be a little trickier for `diversion`, because `diversion` uses dns commands to filter ads, which means that the new `dnsmasq` may make `diversion` no longer work.

JACK-THINK, 2020-05-16:
@PC-Pilot @JackMerlin 

I checked first 1000 lines of source code of **Skynet**. It seems that a lot of unique features of **ASUS-MERLIN** are necessary, such as `/jffs/scripts/firewall-start`, `/jffs/scripts/post-mount`, `/jffs/scripts/unmount` and **Skynet** covers a lot of things beyond functions of firewall like `swap`, `webui` and others. In this case, sheer amount of modification must be done to the source of **Skynet** in order to integrate it in my project, since I have to ensure that it can run perfectly on both **ASUSWRT and ASUS-MERLIN**.

I've also sent an invitation to the developer of **Skynet** asking for help to accomplish this goal, as you see above. Help from him is apparently dispensable.

JackMerlin, 2020-05-16:
> I checked first 1000 lines of source code of **Skynet**. It seems that a lot of unique features of **ASUS-MERLIN** are necessary, such as `/jffs/scripts/firewall-start`, `/jffs/scripts/post-mount`, [...]

I mean to improve the compatibility of your scripts with skynet in merlin firmware, I think skynet is difficult to support stock firmware, because many commands are not supported in stock firmware, like `curl` and advanced `gerp` options, Of course, you can solve this problem by installing updated binaries, but the new binaries are not system paths.

And you also know that skynet uses a lot of Merlin firmware trigger scripts, like `firewall-start`. for example, every time the firewall is restarted, all `iptables` and `ipset` rules will be deleted, which is the stock firmware feature, but the Merlin firmware allows you to run a script after the firewall is started, so you can add your rules again.

So it is very difficult to implement the monitoring firewall startup on the stock firmware, you will need a background script, which may consume a lot of system resources. Because the Merlin firmware monitors the firewall in the `watchdog`, which is part of the `rc`, it can reduce a lot of system resources, but you cannot modify the `rc` of the stock firmware unless you rebuild a firmware.

So, I think we don't need to add skynet to stock firmware, but we can improve the compatibility of skynet and your scripts in Merlin firmware.

JACK-THINK, 2020-05-16:
@JackMerlin 

Get it! We have the same view that it can be very difficult to make **Skynet** run perfectly on ASUSWRT since the difference caused by `rc`. But is it possible to use `monit` to replace `rc`?

As to the requisite commands like `grep` or `curl`, they're easy to get from Entware.

To solve the problem of path, just adjust the path to these commands, like adding `/opt/bin/`, which should be easy.

To make **Skynet** compatible with my project, we also need to remove or disable all the code from **Skynet** except the part which only covers iptables and ipset, since things like swap from **Skynet** will definitely collide with swap from my project.

As stated above, sheer amount of modification must be done to the source of **Skynet**.

JackMerlin, 2020-05-16:
@JACK-THINK
> But is it possible to replace `rc` with `monit`?

Sorry i didn't know `monit` before,  read wikipedia just a few minutes ago, I think the `monit` might be competent for this job.

But I suggest not to add skynet to stock firmware. the reason is very simple. skynet is the most complicated user script in the merlin firmware. If you change the whole skynet to be compatible with your scripts, it may be very very difficult.
So my suggestion is to make skynet compatible with your scripts in the merlin firmware. Do not change any skynet script, but change your scripts for compatibility.

For example, if your scripts detects that the user has skynet installed, try not to create a new `swap` again. this should be easy, I remember that amtm also supports `swap`, but it works well with skynet. Just check the [source code of amtm](https://github.com/decoderman/amtm), I think it will be very helpful.

JACK-THINK, 2020-05-16:
@JackMerlin 

OK. I'll give it a try, but at this point, I just can not guarantee anything. And I need a few days to read each line of the source code of `Skynet` first and make a feasible solution.

One more thing, are you the author of this [post](http://www.52asus.com/thread-3819-1-1.html), the **JACK-** I know long time ago?

JackMerlin, 2020-05-16:
@JACK-THINK 
> OK. I'll give it a try, but at this point, I just can not guarantee anything. And I need a few days to read each line of the source code of `Skynet` first and make a feasible solution.

Try to flash the merlin firmware to your router, and install your scripts and then install the skynet, if anything doesn't work try to modify your scripts.

Skynet is very complicated for me, and it took me several hours to understand one line of code. So I believe this is a difficult job, thank you for trying to be compatible with it!

But the advantage of reading the code is that it will bring a lot of new ideas to your project. for example, when I write a USB Accelerator, I need to automatically update from github, but if all users update at the same time that will be like a DDoS attack for github, so I read the skynet code and [used random minutes to saved github](https://github.com/JackMerlin/USBAccelerator/blob/v2.0/usbaccelerator.sh#L3193). 😁

@Adamm00 's scripts are very powerful, and many of them can be learned!

> One more thing, are you the author of this [post](http://www.52asus.com/thread-3819-1-1.html), the **JACK-** I know long time ago?

Yes, they are all my accounts, and sometimes I will post an exclusive thread at 52asus, because I have a good relationship with the administrator there.

PC-Pilot, 2020-05-16:
Jack, I would like to applaud you for investigating the possibilities for making compatible alongside/integrating Skynet and wish you every success in this quest as I feel sure that your versitile SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER would be even more popular especially for novice users of the Asus Merlin firmware like myself!

I see many posts on the excellent https://www.snbforums.com/forums/asuswrt-merlin.42/ resource were individuals new to these advanced router features would eagerly seize upon a reliable automated installer :) !

Best wishes and thank you again for your interest and enthusiastic efforts...

PC Pilot.

JACK-THINK, 2020-05-16:
@JackMerlin 

Thanks for your helpful suggestion.

One more thing, I do really appreciate your feasible solution to monitor the temperature of the router. I have already added a brand new add-on called "[thermometer](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/script_bootloader/usr/thermometer/README_zh-CN.md)" in my project using your solution. It runs perfect on both ASUSWRT and ASUS-MERLIN. Thanks a lot.

By the way, do you still remember the discussion we had about fan-control? I made it work (still in beta status) the day you denied the plan.😂 What's more, you can use **any conditional statement** to make the fan rotate or stop **without any interference on the plug-in USB flash drive**. Because of your reasonable concern, I won't publish it. If you want to read it, leave your e-mail address here and I'll send you the source code.

JackMerlin, 2020-05-16:
@PC-Pilot 
> edit add to that @JackMerlin too ….for collaboration is frequently the key to success!!

Thank you.

@JACK-THINK
> Thanks for your helpful suggestion.

You're welcome.

> One more thing, I do really appreciate your feasible solution to monitor the temperature of the router. I have already added a brand new add-on called [...]

I subscribed to your code repository, so I will receive messages when there are changes. I have seen that script when you release it. that script is very great. actually, I never thought of using `nvram get wl_ifnames`  to get the temperature of the wireless chip, this is definitely a very great idea. so I am very honored to be able to help you.

> By the way, do you still remember the discussion we had about fan-control? I made it work (still in beta status) the day you denied the plan.😂 What's more, you can use **any conditional statement** [...]

This is very helpful. In fact, I don't dare to use a usb fan anymore. I use the power adapter of my phone. Although I have no way to control its power, but I feel not bad.
If you are willing to share your experience, I definitely SNB is a good forum, most people in SNB know the consequences of doing so, if you have enough warnings, I think there is no big deal. so I did not close [this](https://www.snbforums.com/threads/control-the-usb-power-supply-on-off-with-commands.62839/) SNB thread, but I won't share it elsewhere because people don't know what they are doing or even what they need.

Thanks both.

JACK-THINK, 2020-05-19:
Because of some well-known reason, this repository won't be updated until July.