JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #62: 安装插件时无法获取正确的URL。

befantasy, 2020-05-30:
所有的addon都无法安装。
以aliyunw为例，url有误，可以看到中间download//aliyun中间少了版本号。其他所有的也都一样。
https://github.com/JACK-THINK/aliyun-cli/releases/download//aliyun-cli-linux-arm.zip

befantasy, 2020-06-08:
这么严重的问题作者都不管吗？？？
插件一个都安装不了。

befantasy, 2020-06-08:
看了一下v2ray addon的安装脚本
/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/v2ray/bin/v2ray_install

脚本中的下载连接为

` DOWNLOAD_LINK="https://github.com/v2ray/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`

执行脚本看到的报错

>  https://github.com/v2ray/v2ray-core/releases/download//v2ray-linux-arm.zip
> Resolving github.com... 13.229.188.59
> Connecting to github.com|13.229.188.59|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2020-06-08 14:15:21 ERROR 404: Not Found.
> 

可以看到url中${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE的变量都正常解析出来了，而${LATEST_VERSION}变量没有正常解析。导致404 not found。

继续查看安装脚本，${LATEST_VERSION}变量

` LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`

单独执行

> /opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" 

返回完全正常。正常返回了v2ray的所有latest release信息。

至此，基本判定作者的sed正则表达式写的有问题，或者是v2ray的release命名方式改了，导致无法解析latest version信息，安装失败。不过我对正则表达式一窍不通，所以也无法提交PR。

搜了好几个issues，都有类似的问题，不但v2ray无法安装，我试了好几个addon都无法安装。我觉得作者的回答十分不负责任，全都看都不看，一概推给github被污染。。。

结合项目的安装过程中，U盘分区，entware安装等过程出过好多莫名其妙的问题。总体感觉这个项目很不靠谱，本来寄予厚望，结果失望至极。

JackMerlin, 2020-06-22:
@befantasy @JACK-THINK 
无需那么复杂的正则表达式
```shell
curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | grep -i "tag_name" | awk -F'"' '{print $4}'
```

edwinhuish, 2020-06-23:
@JACK-THINK 本来想 git clone 下来测试一下这个问题，但是git clone了一天都下不来。通过美国VPS下下来检查发现 .git 下面有个 .pack 文件非常大。建议你按照这个知乎整理一下 git。

另外，@JackMerlin 的解决方案我测试了一下，结果非常诡异，如下图：

[image]

有时候能够正常，有时候却返回莫名其妙的字符串。除了 github 在大陆网络不稳定之外，我暂时想不到其他可能性。。

edwinhuish, 2020-06-23:
> @JACK-THINK 本来想 git clone 下来测试一下这个问题，但是git clone了一天都下不来。通过美国VPS下下来检查发现 .git 下面有个 .pack 文件非常大。建议你按照这个知乎整理一下 git。
> 
> 另外，@JackMerlin 的解决方案我测试了一下，结果非常诡异，如下图：
> 
> [image]
> 
> 有时候能够正常，有时候却返回莫名其妙的字符串。除了 github 在大陆网络不稳定之外，我暂时想不到其他可能性。。

@JackMerlin 这个诡异的问题找到原因了，GitHub api 的返回有时候是不能正常识别为Json，而是没有换行的一堆字符串，所以你这个方式不行。。

edwinhuish, 2020-06-23:
研究了一下 shell 的正则，并经过测试。下面这个方式可以完美匹配，无论是有换行的json，还是被压缩过的字符串。

```
/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | grep -i -o -E \"tag_name\":[[:blank:]]*\"[^\"]+ | awk -F'"' '{print $4}'
```
@JACK-THINK  希望能修正，PR 就不提了，800多MB 的git 怕了。

JACK-THINK, 2020-08-09:
@be

> 看了一下v2ray addon的安装脚本
> /tmp/mnt/ASUS_ROUTER/script_bootloader/usr/v2ray/bin/v2ray_install
>
> 脚本中的下载连接为
>
> ` [...]

升米恩斗米仇是吧？你在墙内访问Github被污染+免费版Github API访问频率限制，导致不能返回版本号，这也是我的锅？？？这套代码在RT-AC68U和RT-AC86U平台上是100%通过测试的。还有，这是一个纯开源免费的项目，你居然要求作者负责？？知道什么是GPLv2协议吗？爱用用，不用滚！用不着你在这里说三道四

closed 2020-08-09 by JACK-THINK

befantasy, 2020-08-09:
我就问问别人提的issues你回复之前都看了没试了没？ 你这样的人，既然别人提的问题你看都看不进去，把代码放到github上干什么？自己存本机上自嗨不就完了？

能上github的人不知道github抽风？不会爬个墙？我说了：
```
/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest"
单独执行curl返回完全正常。正常返回了v2ray的所有latest release信息。
```

上面有多少人说了，你正则表达式写的有瑕疵。我要求你对项目负责了吗？我说的是你回复issue负点责任，至少看看、试试，别看都不看就说是github的问题。哪怕你不回都没问题。