JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #69: 在线安装和离线安装都出现错误：CAN'T RE-PARTITION Kingston DataTraveler 3.0

xiaosan0318, 2020-08-11:
华硕：RT-AC66U_B1
梅林：384.18
U盘：金士顿64G
日志：
admin@RT-AC66U_B1-18A0:/tmp# cd /tmp && wget -q -O /tmp/install_online --no-check-certificate "https://cdn.jsdelivr.net/gh/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER@late
st/script_bootloader/bin/install_online" && chmod 777 /tmp/install_online && /tmp/install_online
############### WARNING ###############
This program will erase ALL THE DATA on the target USB flash drive and install SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER.
Before installation, you must:
1. Back up all the data on the target USB flash drive.
2. Remove all the USB flash drives from the router except the target one.

############### 警告 ###############
本程序将清除目标U盘中全部数据并安装SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER。
开始安装前，请务必：
1. 备份目标U盘上的全部数据。
2. 除目标U盘外，移除全部连接至路由器的U盘。

Do you want to start installation right now? (YES/NO): YES
(install_online): 22583 INFO: START INSTALLATION
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2020-08-12 00:06:03--  https://cdn.jsdelivr.net/gh/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER@latest/script_bootloader/bin/drive_modifier
Resolving cdn.jsdelivr.net... 240e:f7:8e00:100::3e, 36.25.241.35
Connecting to cdn.jsdelivr.net|240e:f7:8e00:100::3e|:443... failed: Network is unreachable.
Connecting to cdn.jsdelivr.net|36.25.241.35|:443... connected.
HTTP request sent, awaiting response... 416 Requested Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

(drive_modifier): 22812 INFO: RE-PARTITION Kingston DataTraveler 3.0

The number of cylinders for this disk is set to 4743.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Changing display/entry units to sectors

Command (m for help):
Disk /dev/sda: 30.9 GB, 30943995904 bytes
245 heads, 52 sectors/track, 4743 cylinders, total 60437492 sectors
Units = sectors of 1 * 512 = 512 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1   ?  4109694196  3924421095  2054847098  f4 Unknown
/dev/sda2   ?  4109694196  3924421095  2054847098  f4 Unknown
/dev/sda3   ?  4109694196  3924421095  2054847098  f4 Unknown
/dev/sda4   ?  4109694196  3924421095  2054847098  f4 Unknown

Command (m for help): Building a new DOS disklabel. Changes will remain in memory only,
until you decide to write them. After that the previous content
won't be recoverable.

The number of cylinders for this disk is set to 4743.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Command action
   e   extended
   p   primary partition (1-4)
Partition number (1-4): First sector (52-60437491, default 52): Last sector or +size or +sizeM or +sizeK (2048-60437491, default 60437491): Using default value 60437491

Command (m for help):
Disk /dev/sda: 30.9 GB, 30943995904 bytes
245 heads, 52 sectors/track, 4743 cylinders, total 60437492 sectors
Units = sectors of 1 * 512 = 512 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1            2048    60437491    30217722  83 Linux

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table
**(drive_modifier): 22812 NOTICE: CAN'T RE-PARTITION Kingston DataTraveler 3.0**
**(install_online): 22583 FAILURE: ONLINE-INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER**

手动安装：
admin@RT-AC66U_B1-18A0:/tmp# /tmp/drive_modifier
(drive_modifier): 18406 INFO: RE-PARTITION Kingston DataTraveler 3.0

The number of cylinders for this disk is set to 4743.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Changing display/entry units to sectors

Command (m for help):
Disk /dev/sda: 30.9 GB, 30943995904 bytes
245 heads, 52 sectors/track, 4743 cylinders, total 60437492 sectors
Units = sectors of 1 * 512 = 512 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1   ?  4109694196  3924421095  2054847098  f4 Unknown
/dev/sda2   ?  4109694196  3924421095  2054847098  f4 Unknown
/dev/sda3   ?  4109694196  3924421095  2054847098  f4 Unknown
/dev/sda4   ?  4109694196  3924421095  2054847098  f4 Unknown

Command (m for help): Building a new DOS disklabel. Changes will remain in memory only,
until you decide to write them. After that the previous content
won't be recoverable.

The number of cylinders for this disk is set to 4743.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Command action
   e   extended
   p   primary partition (1-4)
Partition number (1-4): First sector (52-60437491, default 52): Last sector or +size or +sizeM or +sizeK (2048-60437491, default 60437491): Using default value 60437491

Command (m for help):
Disk /dev/sda: 30.9 GB, 30943995904 bytes
245 heads, 52 sectors/track, 4743 cylinders, total 60437492 sectors
Units = sectors of 1 * 512 = 512 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1            2048    60437491    30217722  83 Linux

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table
**(drive_modifier): 18406 NOTICE: CAN'T RE-PARTITION Kingston DataTraveler 3.0**

没有script_bootloader目录
admin@RT-AC66U_B1-18A0:/tmp# /tmp/mnt/ASUS_ROUTER/script_bootloader/bin/install
-sh: /tmp/mnt/ASUS_ROUTER/script_bootloader/bin/install: not found

admin@RT-AC66U_B1-18A0:/tmp# ls /tmp/mnt/ASUS_ROUTER/
bin/  etc/  tmp/  usr/  var/

尝试安装/tmp/mnt/ASUS_ROUTER/bin/install也不行。
admin@RT-AC66U_B1-18A0:/tmp# ls /tmp/mnt/ASUS_ROUTER/bin/install
install         install_online
admin@RT-AC66U_B1-18A0:/tmp# /tmp/mnt/ASUS_ROUTER/bin/install
(install): 15432 *--------- INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER ON RT-AC68U ---------*
(install): 15432 NOTICE: UNABLE TO LOCATE DIRECTORY script_bootloader
(install): 15432 FAILURE: INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER

xiaosan0318, 2020-08-11:
admin@RT-AC66U_B1-18A0:/tmp# cat /tmp/usb.log
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.
[usb_dbg: get_usb_node_by_device] (lp0): Fail to get link: /sys/class/usb/lp0/device.
[usb_dbg: get_usb_node_by_device] (lp1): Fail to get link: /sys/class/usb/lp1/device.

[image]

voidiak, 2021-08-14:
您好，我也遇到了这个问题，请问您后来解决这个问题了么？