JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #75: 金士顿32G U盘分区失败

yuppiesnotzhuhao, 2020-10-23:
admin@RT-AX86U-0510:/tmp# cd /tmp && wget -q -O /tmp/install_online --no-check-certificate "https://cdn.jsd
elivr.net/gh/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER@latest/script_bootloader/bin/install_online" && 
chmod 777 /tmp/install_online && /tmp/install_online
############### WARNING ###############
This program will erase ALL THE DATA on the target USB flash drive and install SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER.
Before installation, you must:
1. Back up all the data on the target USB flash drive.
2. Remove all the USB flash drives from the router except the target one.

############### 警告 ###############
本程序将清除目标U盘中全部数据并安装SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER。
开始安装前，请务必：
1. 备份目标U盘上的全部数据。
2. 除目标U盘外，移除全部连接至路由器的U盘。

Do you want to start installation right now? (YES/NO): YES
(install_online): 29939 INFO: START INSTALLATION
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/root/.wget-hsts'. HSTS will be disabled.
--2020-10-23 22:15:22--  https://cdn.jsdelivr.net/gh/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER@latest/script_bootloader/bin/drive_modifier
Resolving cdn.jsdelivr.net... 219.144.91.65, 240e:bf:b800:100::8
Connecting to cdn.jsdelivr.net|219.144.91.65|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5879 (5.7K) [application/octet-stream]
Saving to: '/tmp/drive_modifier'

/tmp/drive_modifier        100%[=======================================>]   5.74K  --.-KB/s    in 0s      

2020-10-23 22:15:22 (627 MB/s) - '/tmp/drive_modifier' saved [5879/5879]

(drive_modifier): 30164 INFO: RE-PARTITION Kingston DataTraveler 3.0 PMAP

The number of cylinders for this disk is set to 29592.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Changing display/entry units to sectors

Command (m for help): 
Disk /dev/sda: 31.0 GB, 31029460992 bytes
64 heads, 32 sectors/track, 29592 cylinders, total 60604416 sectors
Units = sectors of 1 * 512 = 512 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1   ?     6579571  1924427647   958924038+ 70 Unknown
Partition 1 does not end on cylinder boundary
/dev/sda2   ?  1953251627  3771827541   909287957+ 43 Unknown
Partition 2 does not end on cylinder boundary
/dev/sda3   ?   225735265   225735274           5  72 Unknown
Partition 3 does not end on cylinder boundary
/dev/sda4      2642411520  2642463409       25945   0 Empty
Partition 4 does not end on cylinder boundary

Partition table entries are not in disk order

Command (m for help): Building a new DOS disklabel. Changes will remain in memory only,
until you decide to write them. After that the previous content
won't be recoverable.

The number of cylinders for this disk is set to 29592.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Command action
   e   extended
   p   primary partition (1-4)
Partition number (1-4): First sector (32-60604415, default 32): Last sector or +size or +sizeM or +sizeK (2048-60604415, default 60604415): Using default value 60604415

Command (m for help): 
Disk /dev/sda: 31.0 GB, 31029460992 bytes
64 heads, 32 sectors/track, 29592 cylinders, total 60604416 sectors
Units = sectors of 1 * 512 = 512 bytes

   Device Boot      Start         End      Blocks  Id System
/dev/sda1            2048    60604415    30301184  83 Linux

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table
(drive_modifier): 30164 NOTICE: CAN'T RE-PARTITION Kingston DataTraveler 3.0 PMAP
(install_online): 29939 FAILURE: ONLINE-INSTALL SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER

leechaojie, 2021-03-04:
我刚开始也遇到这种问题，我的U盘是 NTFS 格式，格式化成 FAT 后问题解决

leechaojie, 2021-03-04:
FAT格式化后路由器重启不会自动识别，只能自动识别 NTFS，否则就得重新插拔，换了个陈年老旧的的8G usb3.0 的 u盘后顺利上车。

deechoe, 2021-03-13:
Make sure you're using the USB 2.0 port.  
The 3.0 port has been known to have problems.  See if that works.

yuppiesnotzhuhao, 2021-03-13:
> Make sure you're using the USB 2.0 port.
> The 3.0 port has been known to have problems. See if that works.

Thanks for reply,and I had already change another USB flash drive to solve this problem.

closed 2021-03-13 by yuppiesnotzhuhao