JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #79: V2ray下载地址失效

YuMXuan, 2020-12-16:
麻烦更新一下V2ray下载地址。

YuMXuan, 2020-12-16:
可能是因为版本老旧，因为网络不好下的离线安装包，发现是17.0版本。
等更新新版本之后再反馈。

YuMXuan, 2020-12-16:
经过检查，v2ray账号现已换为v2fly，以下代码：
`DOWNLOAD_LINK="https://github.com/v2ray/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`
及，
`LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`
应更改为：
`DOWNLOAD_LINK="https://github.com/v2fly/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`
及，
`LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2fly/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`

需要测试可行性，直连网络不通畅···

dyytian, 2021-01-02:
感谢，看到这个总算是装上了

whave, 2021-02-12:
> 经过检查，v2ray账号现已换为v2fly，v2ray_install中以下代码：
> `DOWNLOAD_LINK="https://github.com/v2ray/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`
> 及，
> `LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2ray/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`
> 应更改为：
> `DOWNLOAD_LINK="https://github.com/v2fly/v2ray-core/releases/download/${LATEST_VERSION}/v2ray-${OPERATING_SYSTEM}-${ROUTER_ARCHITECTURE}.zip"`
> 及，
> `LATEST_VERSION="$(/opt/bin/curl -s "https://api.github.com/repos/v2fly/v2ray-core/releases/latest" | /opt/bin/sed -E -n -e 's/^[[:space:]]*\"tag_name\":[[:space:]]*\"(.*)\".*/\1/p')"`
> 
> 需要测试可行性，直连网络不通畅···

请问一下，安装好V2RAY后，提示输入transport settings ,只有TCP和HTTPS两个选项，可是V2RAY一般用的都是ws传输协议，请问这个怎么解决啊

whave, 2021-02-12:
请问大大，V2RAY有像adguardhome （Web页面登陆地址：http://<路由器IP地址>:14711）一样的WEB登录配置界面吗