JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #85: Version 20.0&18.5 INSTALL ADGUARD-HOME FAILURE

593975101, 2021-03-25:
(adguardhome_install): 5222 *--------- INSTALL ADGUARD-HOME ON RT-AC86U ---------*
(adguardhome_install): 5222 INFO: ENTWARE_ENABLE SERVICE HAS BEEN LOADED
(adguardhome_install): 5222 INFO: MONIT MANAGEMENT SYSTEM HAS BEEN DEPLOYED
/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/adguardhome/bin/adguardhome_install: line 228: /opt/bin/wget-ssl: not found
(adguardhome_install): 5222 NOTICE: CONNECTION FAILED
(adguardhome_install): 5222 FAILURE: INSTALL ADGUARD-HOME

593975101, 2021-03-27:
ASUSWRT-Merlin RT-AC86U 386.1_2 Fri Feb 12 22:48:22 UTC 2021
RT-AC86U@RT-AC86U-B050:/tmp/home/root# cd /tmp && /tmp/mnt/ASUS_ROUTER/script_bootloader/bin/update

############### DATE: Sat Mar 27 13:07:27 CST 2021 ###############

PART 1: SYSTEM

/tmp/mnt/ASUS_ROUTER/script_bootloader/bin/update: line 211: /opt/bin/wget-ssl: not found
FAILURE: CONNECTION FAILED

############### END ###############

leechaojie, 2023-03-12:
同样的问题

leechaojie, 2023-03-13:
进入 `/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/adguardhome/bin` 目录
`vim adguardhome_install` 将 149 行的 ` /opt/bin/wget-ssl` 改成  `/opt/bin/wget` 就可以