JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #86: AdguardHome 客户端显示127.0.0.1

shugo-chara, 2021-05-15:
华硕原版系统，安装adguardhome后服务正常，只是客户端全部显示为127.0.0.1
查找原因说要改adh的设置为重定向为53端口，但路由里没有adh相关选项，adh管理页面上也没有。。

edwinhuish, 2021-05-19:
请看： https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER-ADDONS/blob/master/adguardhome/README_zh-CN.md

[image]

wangdefa8, 2021-05-24:
https://www.right.com.cn/forum/thread-4112404-1-1.html
https://www.right.com.cn/forum/thread-4053020-1-1.html
按照这两个教程设置，可以显示客户端
[image]

edwinhuish, 2021-05-24:
按照 @wangdefa8  的提示，其实可以这样做：
删除或注释 script_bootloader/usr/adguardhome/bin/adguardhome_enable.service 的 79-81行，并添加iptables转发
```
    # /opt/bin/mv -f "/tmp/resolv.dnsmasq" "/tmp/resolv.dnsmasq.bak"
    # /opt/bin/cp -lf "${PRIVATE_ETC}/dnsmasq.d/adguardhome.conf" "${ENTWARE_DNSMASQD}"
    # /opt/bin/dnsmasq.d_refresh > /dev/null 2>&1

    LAN_IP="$(nvram get lan_ipaddr)"
    iptables -t nat -A PREROUTING -p tcp -d "${LAN_IP}" --dport 53 -j REDIRECT --to-port 2153
    iptables -t nat -A PREROUTING -p udp -d "${LAN_IP}" --dport 53 -j REDIRECT --to-port 2153
```

删除或注释  script_bootloader/usr/adguardhome/bin/adguardhome_disable.service 对应的 53-55 行：
```
    # /opt/bin/mv -f "/tmp/resolv.dnsmasq.bak" "/tmp/resolv.dnsmasq"
    # /opt/bin/rm -f "${ENTWARE_DNSMASQD}/adguardhome.conf"
    # /opt/bin/dnsmasq.d_refresh > /dev/null 2>&1
```

wangdefa8, 2021-05-24:
@edwinhuish 我对这方面是个小白，我主要是按照上面两个链接来设置的，因为之前用的是老毛子固件，可以
显示客户端。
梅林的话我是按照作者的提示这样做的：
1.winscp进入/opt/etc/dnsmasq.d，新建dhcp_option_125.conf，内容为port=0，权限0755（权限这步好像保持默认也可以，具体没有验证）
2.进入/jffs/scripts，新建firewall-start文件，权限0755（这个必须要改）内容为
#!/bin/sh
iptables -t nat -A PREROUTING -p tcp -d 192.168.123.1 --dport 53 -j REDIRECT --to-ports 2153
iptables -t nat -A PREROUTING -p udp -d 192.168.123.1 --dport 53 -j REDIRECT --to-ports 2153
192.168.123.1是我的路由器后台，可以改为你的路由器后台地址。2153是adh的监控端口，这个不建议改，改了好像会出错。
重启路由器，进ad后台，应该就可以显示客户端了，但是是显示ip地址，要显示设备名的话就在adh的设置——客户端设置——添加客户端，根据自己的情况添加，就好了。

shugo-chara, 2021-09-21:
> 按照 @wangdefa8 的提示，其实可以这样做：
> 删除或注释 script_bootloader/usr/adguardhome/bin/adguardhome_enable.service 的 79-81行，并添加iptables转发
> 
> ```
>     # /opt/bin/mv -f "/tmp/resolv.dnsmasq" "/tmp/resolv.dnsmasq.bak"
>     # /opt/bin/cp -lf "${PRIVATE_ETC}/dnsmasq.d/adguardhome.conf" "${ENTWARE_DNSMASQD}"
>     # /opt/bin/dnsmasq.d_refresh > /dev/null 2>&1
> 
>     LAN_IP="$(nvram get lan_ipaddr)"
>     iptables -t nat -A PREROUTING -p tcp -d "${LAN_IP}" --dport 53 -j REDIRECT --to-port 2153
>     iptables -t nat -A PREROUTING -p udp -d "${LAN_IP}" --dport 53 -j REDIRECT --to-port 2153
> ```
> 
> 删除或注释 script_bootloader/usr/adguardhome/bin/adguardhome_disable.service 对应的 53-55 行：
> 
> ```
>     # /opt/bin/mv -f "/tmp/resolv.dnsmasq.bak" "/tmp/resolv.dnsmasq"
>     # /opt/bin/rm -f "${ENTWARE_DNSMASQD}/adguardhome.conf"
>     # /opt/bin/dnsmasq.d_refresh > /dev/null 2>&1
> ```
> 
> 不过这样做有个弊端，禁用 Adguard Home 的时候，没有同时删除iptables，需要重启。因为我不知道怎么通过shell去删除。

感谢，按你所说顺利显示客户端ip了。不过不知道为什么一直把最多访问的127.0.0.1识别为我的电脑ip，而单独给电脑ip设置客户端并不显示流量，别的一切正常。

shugo-chara, 2021-09-21:
> @edwinhuish 我对这方面是个小白，我主要是按照上面两个链接来设置的，因为之前用的是老毛子固件，可以
> 显示客户端。
> 梅林的话我是按照作者的提示这样做的：
> 1.winscp进入/opt/etc/dnsmasq.d，新建dhcp_option_125.conf，内容为port=0，权限0755（权限这步好像保持默认也可以，具体没有验证） [...]

感谢。SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER项目把对jffs的写入等操作全部重写了，试了一下你的方法会无法无法联网- -最后用楼上的直接修改adh的服务参数来变通解决了。

kunki, 2023-05-21:
> 按照 @wangdefa8 的提示，其实可以这样做： 删除或注释 script_bootloader/usr/adguardhome/bin/adguardhome_enable.service 的 79-81行，并添加iptables转发
>
> ```
>     # /opt/bin/mv -f "/tmp/resolv.dnsmasq" "/tmp/resolv.dnsmasq.bak" [...]

刪除iptables可以用-D，完整的指令爲：
```
    # /opt/bin/mv -f "/tmp/resolv.dnsmasq.bak" "/tmp/resolv.dnsmasq"
    # /opt/bin/rm -f "${ENTWARE_DNSMASQD}/adguardhome.conf"
    # /opt/bin/dnsmasq.d_refresh > /dev/null 2>&1
    LAN_IP="$(nvram get lan_ipaddr)"
    iptables -t nat -D PREROUTING -p tcp -d "${LAN_IP}" --dport 53 -j REDIRECT --to-port 2153
    iptables -t nat -D PREROUTING -p udp -d "${LAN_IP}" --dport 53 -j REDIRECT --to-port 2153
```
另外可用```iptables -L --t nat```查看當前路由表，驗證刪除效果