JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #91: On Stage 2, it doesn´t move at all

AZBoxEmperador, 2021-08-23:
[image]

What am suppose to do here?, please confirm it has been like this for more than 3 hours, I'm currently on asus firmware: 3.0.0.4.386_43129 and using a: RT-N68U

Maasea, 2021-09-06:
This is a bug when you use the online-install script, it should  show you a tips to  input size twice.
For another word, you should input the size  then enter  and reinput it then enter

liuwangdao, 2022-03-13:
your usb device maybe is in bad status 
close your router for a few hours or just wait for a few hour then run
/tmp/mnt/ASUS_ROUTER/script_bootloader/bin/install
to retry