JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #104: 路由器重启后无法加载环境

bytxk, 2023-08-02:
AX88U PRO 最新固件[3.0.0.6.102_21514]离线安装v26.0&18.11版本后可正常使用如monit aria2，重启后无法加载环境。script_bootloader_usb_mount等都是有执行权限的。

bytxk, 2023-08-02:
[直接执行/tmp/mnt/ASUS_ROUTER//script_bootloader/bin/script_bootloader_usb_mount可挂载，但是avah加载失败[image]

bytxk, 2023-08-02:
官方固件执行nvram show已经没有script_mount和script_umount参数了

Young-in-the-middle, 2023-11-01:
同样遇到这个问题。 还是的用jffs的post-mount 才行。

JACK-THINK, 2023-12-17:
这里显示的avahi加载失败是正常的，因为它是由Entware安装的，不能让它正常运行，因为可能与路由器内置的avahi服务程序冲突。通过`ps | grep 'avahi'`可以看到路由器固件内置的avahi已经正常运行

JACK-THINK, 2023-12-17:
> AX88U PRO 最新固件[3.0.0.6.102_21514]离线安装v26.0&18.11版本后可正常使用如monit aria2，重启后无法加载环境。script_bootloader_usb_mount等都是有执行权限的。

重启无法加载可能是由于[这个原因](https://github.com/JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER/blob/master/ChangeLog_zh-CN.md#%E7%89%88%E6%9C%AC2802023-12-16)。