JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #106: AX86U 等型号更新固件后 script_usbmount 开机启动失效的解决方案

mlkt, 2023-12-19:
研究了一下怎么解决，在这做个记录。

先安装下载大师，USB相关应用 -> 下载大师，点击安装：
[image]

进入ssh，执行以下命令，修改下载大师的启动脚本：
```
vi $(nvram get apps_mounted_path)/$(nvram get apps_install_folder)/etc/init.d/S50downloadmaster
```

在开头加入以下代码：
```
{
    if [ "$1" != "start" ] && [ "$1" != "restart" ]; then
        exit
    fi
    local SCRIPT_PATH=`nvram get script_usbmount`
    if [ ! "$SCRIPT_PATH" ]; then
        exit
    fi
    local ONCE_TMP_FILE=/tmp/script_usbmount_executed
    if [ -f "$ONCE_TMP_FILE" ]; then
        exit
    fi

    local APPS_MOUNTED_PATH=`nvram get apps_mounted_path`
    sh -c "\"$SCRIPT_PATH\" \"$APPS_MOUNTED_PATH\"" >/dev/null 2>&1 || true

    touch "$ONCE_TMP_FILE"
} &
```

变成类似这样：
[image]

这样系统启动下载大师的时候就会去运行之前在 script_usbmount 配置项里设置的脚本，和之前的执行方式一样，且只会在开机的时候运行一次，不管怎么启停下载大师都不会重复运行，也不会影响下载大师的功能。注意如果更新下载大师，这个启动脚本有可能会被还原，需要再修改一次。

如果要兼容支持/不支持 script_usbmount 配置项的固件，需要再修改 script_usbmount 里指定的启动脚本的文件内容，在开头加上：
```
{
    local ONCE_TMP_FILE=/tmp/script_usbmount_executed
    if [ -f "$ONCE_TMP_FILE" ]; then
        exit
    fi
    touch "$ONCE_TMP_FILE"
}
```
这样不管是不是支持 script_usbmount ，都只会在开机时执行一次这个脚本。

JACK-THINK, 2024-01-14:
感谢反馈。但是问题无法按此方式简单解决：

您提到的方法是通过调用固件源码[usb.c](https://github.com/RMerl/asuswrt-merlin.ng/blob/1de00d63c8594a9ab7520808905c8be4c97e9d74/release/src/router/rc/usb.c)模块，依次执行
1. `/usr/sbin/app_check_folder.sh` # 检查外置磁盘挂载点下是否存在合法asusware目录
2. `/usr/sbin/app_fsck.sh` # 检查磁盘错误
3. `rm -rf /tmp/opt` # 取消opt软链接
4. `ln -sf 磁盘挂载点/合法asusware目录 /tmp/opt` # 新建opt软链接，指向外置磁盘根目录下合法asusware目录
   > 从此开始后续步骤均有可能与本系统冲突
5. `APPS_DEV=sda1 APPS_MOUNTED_PATH=/tmp/mnt/ASUS_ROUTER /rom/.asusrouter` # 设置环境变量`APPS_DEV`与`APPS_MOUNTED_PATH`，执行`/rom/.asusrouter`
   1. `/usr/sbin/app_base_link.sh` # 创建`/opt/*/*`软链接至`/tmp/mnt/ASUS_ROUTER/合法asusware目录/*/*`
      > 本系统内含Entware，亦依赖`/opt/*/*`，因此，极有可能存在严重冲突
   2. 创建并启用swap
   3. `/usr/sbin/app_update.sh` # 更新apps
   4. `/usr/sbin/app_init_run.sh allpkg start` # 启动全部apps
      > 此文件负责全部`/opt/etc/init.d/S*`文件的`start`与`stop`，与Entware的引导文件`rc.unslung`功能重叠。而且，通过Entware安装的程序，也将init文件保存在`/opt/etc/init.d`目录中，可视为严重冲突。如采用此方法，则必须完全重写Entware的引导文件`rc.unslung`，使之与`/usr/sbin/app_init_run.sh`功能互补，加之`/usr`目录下文件均为固件内容，完全不可更改，因此很有难度。即使适配成功，只要后续华硕原厂固件对当前引导方式做出修改，就有可能引入**无法解决**的冲突，风险很大。本次，华硕官方将存在了**至少12年**的`script_usbmount`调用方法移除，足以为证。

综上，转用梅林固件，使用它提供的**不可能删除的`/jffs/scripts`方法**调用本系统，才是**完全无冲突**且**一劳永逸**的解决办法。

mlkt, 2024-01-18:
理解，这个方法太hack了，确实是不适合作为你这个项目内的解决方案，我只是想在这记录一下这个配置项失效后开机启动的变通方式，因为很多情况可能只是想启动一两个自己的程序而已，没有必要搞太复杂。

JACK-THINK, 2024-01-19:
其实这个方法确实是一条路，只是不知道它还能存在多久。如果它能长期保持不变，即使多费点儿力做适配也还算值得。但是，如果华硕将来以**增强安全**为理由，直接把`/usr/sbin/app_init_run.sh`改为只能启动华硕指定的程序并用**sha256校验**来确认相关程序合法性，可理解为白名单机制，那以此为基础的开发将彻底白费，无法挽救。

梅林固件的`/jffs/scripts方法`是固件本身赖以生存的核心功能，具有**不可动摇的稳定性**，尽早转用它，可消除后顾之忧。