JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #109: Entware mirror script

decoderman, 2024-02-24:
Hi JACK-THINK
I'm trying this way to contact you as there is no other way to do so on GitHub.
Would you share your current Entware mirror script with me?
The official, yours and mine are the only Entware repositories I know of. I include all of them in amtm, the addon tool for the official Asuswrt-Merlin firmware.
I would greatly appreciate it.

Best regards, decoderman aka thelonelycoder.

JACK-THINK, 2024-02-28:
@decoderman 

Do you mean the script to build an entware mirror site? This is the shell script.

```sh
#!/bin/sh
#
########## PUBLIC VARIABLES ##########
#
ENTWARE_REPO_HOST="https://bin.entware.net"
ENTWARE_REPO_ROOT="/var/www/ADDRESS.TO.YOUR.ENTWARE.MIRROR.SITE" # DO NOT FORGET TO MODIFY IT
#
########## END ##########
#
#
########## LOAD FUNCTIONS ##########
#
createDirectoryTree()
{
    local TARGET_DIRECTORY="${1}"
    local TEMP_DIRECTORY="${ENTWARE_REPO_ROOT}/sync_tmp"
    local PRESENT_DIRECTORY_TREE="${TEMP_DIRECTORY}/present_directory_tree"
    local LATEST_DIRECTORY_TREE="${TEMP_DIRECTORY}/latest_directory_tree"
    #
    rm -rf "${TEMP_DIRECTORY}"
    #
    mkdir -p "${TEMP_DIRECTORY}"
    #
    wget --timestamping --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --recursive --no-parent --accept "index.html" --directory-prefix="${TEMP_DIRECTORY}" "${ENTWARE_REPO_HOST}"
    #
    find "${TEMP_DIRECTORY}" -type d | sed -E -e '\#/archive/?[^/]*#d' -e 's#'"${TEMP_DIRECTORY}"'#'"${ENTWARE_REPO_ROOT}"'#' -e 's#/$##' > "${LATEST_DIRECTORY_TREE}"
    find "${ENTWARE_REPO_ROOT}" -type d | sed -E -e '\#'"${TEMP_DIRECTORY}"'#d' -e 's#/$##' > "${PRESENT_DIRECTORY_TREE}"
    #
    grep -Fxvf "${PRESENT_DIRECTORY_TREE}" "${LATEST_DIRECTORY_TREE}" | grep -E -e '^'"${TARGET_DIRECTORY}"'' > "${TEMP_DIRECTORY}/list_to_mkdir"
    grep -Fxvf "${LATEST_DIRECTORY_TREE}" "${PRESENT_DIRECTORY_TREE}" | grep -E -e '^'"${TARGET_DIRECTORY}"'' > "${TEMP_DIRECTORY}/list_to_remove"
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_mkdir")
    do
        mkdir -p "${ITEM}" 2> /dev/null
    done
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_remove")
    do
        rm -rf "${ITEM}"
    done
    #
    rm -rf "${TEMP_DIRECTORY}"
}
#
fileSync()
{
    local LOCAL_DIRECTORY="${1}"
    local TEMP_DIRECTORY="${LOCAL_DIRECTORY}/sync_tmp"
    local URL_DIRECTORY="$(echo "${LOCAL_DIRECTORY}" | sed -e 's#'"${ENTWARE_REPO_ROOT}"'/*##')"
    local PRESENT_INDEX="${LOCAL_DIRECTORY}/index.html"
    local CURRENT_INDEX="${TEMP_DIRECTORY}/current_index.html"
    local LATEST_INDEX="${TEMP_DIRECTORY}/latest_index.html"
    #
    rm -rf "${TEMP_DIRECTORY}"
    #
    mkdir -p "${TEMP_DIRECTORY}"
    #
    wget --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --output-document="${LATEST_INDEX}" "${ENTWARE_REPO_HOST}/${URL_DIRECTORY}/"
    #
    if [ -e "${PRESENT_INDEX}" ]
    then
        cp -f "${PRESENT_INDEX}" "${CURRENT_INDEX}"
    else
        echo "" > "${CURRENT_INDEX}"
    fi
    #
    grep -Fxvf "${CURRENT_INDEX}" "${LATEST_INDEX}" | sed -E -n -e 's#^<a href=\"([^/]+)\".*#\1#p' > "${TEMP_DIRECTORY}/list_to_download"
    grep -Fxvf "${LATEST_INDEX}" "${CURRENT_INDEX}" | sed -E -n -e 's#^<a href=\"([^/]+)\".*#\1#p' > "${TEMP_DIRECTORY}/list_to_remove"
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_download")
    do
        wget --timestamping --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --directory-prefix="${LOCAL_DIRECTORY}" "${ENTWARE_REPO_HOST}/${URL_DIRECTORY}/${ITEM}"
    done
    #
    for ITEM in $(cat "${TEMP_DIRECTORY}/list_to_remove")
    do
        rm -f "${LOCAL_DIRECTORY}/${ITEM}"
    done
    #
    cp -f "${LATEST_INDEX}" "${PRESENT_INDEX}"
    #
    rm -rf "${TEMP_DIRECTORY}"
}
#
########## END ##########
#
#
########## MAIN ##########
#
DIRECTORY_TO_SYNC="${ENTWARE_REPO_ROOT}"
#
mkdir -p "${ENTWARE_REPO_ROOT}/css/" 2> /dev/null
wget --no-check-certificate --tries=10 --timeout=100 --limit-rate=5m --continue --no-host-directories --output-document="${ENTWARE_REPO_ROOT}/css/packages.css" "${ENTWARE_REPO_HOST}/css/packages.css"
#
createDirectoryTree "${DIRECTORY_TO_SYNC}"
#
DIRECTORY_LIST="$(find "${DIRECTORY_TO_SYNC}" -type d | sed -e 's#/$##')"
#
for ITEM in ${DIRECTORY_LIST}
do
    fileSync "${ITEM}"
done
#
########## END ##########
```

Run the script with crontab once a week to update your mirror site. And you also need a web server like caddy or nginx to host your mirror site.

decoderman, 2024-02-28:
Thank you so much for the two scripts. I really appreciate it.
My mirror is at https://entware.diversion.ch/ as the need was there when bin.entware.net was unavailable for some. I added your mirror and mine to amtm as a select-able source. It will also auto-switch if one of the three servers fail.

Best regards

closed 2024-02-28 by decoderman