JACK-THINK/SCRIPTS-BOOTLOADER-FOR-ASUS-ROUTER #111: 在线脚本创建swap那里失败。

sqdz0728, 2024-04-27:
[image]
u盘是128G的高速盘，是满足条件的。

sqdz0728, 2024-04-27:
路由器固件版本是：GNUton's Asus Merlin[3004.388.7_1-gnuton0_alpha1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.7_1-gnuton0_alpha1) 
TUF-AX3000_V2_3004_388.7_1-gnuton0_alpha1

roc29a, 2024-08-12:
版本高，往回退

ysbledmh, 2025-10-05:
############### STAGE 2: INSTALL ESSENTIAL COMPONENTS ###############
(swap_install): 5489 *--------- INSTALL SWAP ON RT-AC86U ---------*
mkdir: can't create directory '/script_bootloader/usr/swap/var': No such file or directory
(swap_install): 5489 INFO: CREATE /script_bootloader/usr/swap/var/swapfile.var
/tmp/mnt/ASUS_ROUTER/script_bootloader/usr/swap/bin/swap_install: line 96: /script_bootloader/usr/swap/bin/swap_configure: not found
(swap_install): 5489 NOTICE: /script_bootloader/usr/swap/var/swapfile.var CAN'T BE CREATED
(swap_install): 5489 FAILURE: INSTALL SWAP
(install): 5425 FAILURE: STAGE 2

用同磁碟384版本可正常安裝，386版本不行
用384版本正常後卸載，在更新386後用還原會失敗，修改路徑後能還原但無法進monit
在386版本(還原后)也不能卸載，遺失sb_uninstall腳本